MagicMirrorOrg/MagicMirror #18: Day of the week repeated twice

OhhJared, 2015-04-17:
[image]

The JavaScript is displaying the current temperature as well as the low and high for the day.

fallen293, 2015-04-19:
I noticed that as well today, and it turns out it shows the current day twice because of the day/night forecast breakdown.

OhhJared, 2015-04-19:
@fallen293 Did you find any solutions to the issue?

fallen293, 2015-04-19:
I don't think it is an issue. The first one is the day portion, and the 2nd is the night forecast. Gives you more detail with regards to the day's forecast

OhhJared, 2015-04-19:
True. I wish my JS wasn't awful so I could edit it to ignore that data.

closed 2015-04-22 by OhhJared