MagicMirrorOrg/MagicMirror #26: Face recognition

tomasnorre, 2015-07-31:
Hi, 

I really love the work you have been doing and will try to build my own mirror.

I think it could be cool to extend this with a camera and make face recognition so that my calendar is shown if i'm in front of the mirror, and the calendar of my wife, if it's her in front. 

This would also make it possible to personalise it, as the News i would be interested in would not nesecerily by the same as my wife.. 

+ If a person not recognised is in front of the mirror, it should only display default information.. 

+ If we can make it "intelligent to and check against Facebook, then when friends are walking in front of the mirror.. Hi Peter, You're looking good today. 

Just thoughts, i'm happy to help with implementing my ideas, but first i need to build the mirror.

givre, 2015-07-31:
Hi, 

I will be happy to work on that with people that are interested in this features .
I think the program and all the options about face recognition should become first , then you can build the mirror and integrate the camera ;)

My background is not in this area (face recognition), but I'll be happy to take is seriously and start implementing this. 

Regards, 
Benoit

MichMich, 2015-07-31:
If I recall correctly, there is already someone who did it. I think there is more info about this in the comments of my blog. 

Cheers, Michael

tomasnorre, 2015-07-31:
I want to build the mirror first, as this is a quick win.. But after that i'm into trying to help with this too.. Either adapting something that is already in place, or starting it. 

PS: This is for sure also not my field of expertise, I'm web developer :)

paviro, 2015-12-20:
Got it working :) Give me some time and I will release it but I first have to ask the developer of the code I am heavily using for permission to publish it.

closed 2016-03-30 by MichMich

paviro, 2016-04-03:
In case anyone is still interested who is watching this issue, you can find [MMM-Facial-Recognition](https://github.com/paviro/MMM-Facial-Recognition/) on GitHub now it is usable with the v2 version of the mirror software.