MagicMirrorOrg/MagicMirror #29: Implement smooth weather icons

djsunrise19, 2015-10-19:
Hi,

thank you very much for describing your project. Actually, I'm working on the magic mirror, too :). So, I have a question / an idea:

I saw these very smooth weather icons (https://github.com/maxdow/skycons). Is it possible to implement these?

Greets
Maurice

jusingh2, 2015-10-20:
Dear  Maurice, 
	I believe this is could be implemented. I am not that familiar with JavaScript but at first glance, it looks possible. But it all depends on the end use of the program.

Regards,
Jus Singh

cxdy, 2015-10-20:
Not a developer of the project but I'll take a look tonight and try it out. Should be pretty easy.

djsunrise19, 2015-10-25:
Sounds great. Any new informations here?

Regards
Maurice

cxdy, 2015-10-26:
I've been swamped, apologies. I'll post more when I can.

Dom1n1c, 2015-11-29:
i migrated to a newer version of weathericons, but then the small icons disappear.. maybe i'll find a solution...

closed 2016-02-06 by MichMich