MagicMirrorOrg/MagicMirror #33: Compliments don't work

djsunrise19, 2015-11-02:
Hi,

with the new update, compliments don't change anymore. Anyone else with these problems?

Greets
Maurice

MichMich, 2015-11-03:
I see an issue with the compliments. Will fix it asap. What's the exact issue with the weather forecast?

MichMich, 2015-11-03:
Compliments should be fixed in [this](https://github.com/MichMich/MagicMirror/commit/9a5a74e1538a98bc883f705c5fd8552d7b503575) commit.

djsunrise19, 2015-11-03:
f.e. current temperature on website = 5, on mirror it's 2.7. Or, temparatur min and max are the same on mirror for tomorrow (-4.5), on website it's 2.7min and 7.5 max. perhaps any problems yesterday with the api. today, everything is fine.

thanks for fixing the compliments.

MichMich, 2015-11-04:
@djsunrise19 Cool you please keep an eye on the issue, and open a new issue when it occurs again? Thanks!

closed 2015-11-04 by MichMich