MagicMirrorOrg/MagicMirror #35: New refresh algorithm?

djsunrise19, 2015-11-05:
Hello,

I don't want to sound too petty, but I noticed a disturbing refreshing of the website since the new update. The Version before refreshs each <div> for weather and calendar, in a smooth fx. For me it was great to see, that there is a smooth refresh way per div. NOW the whole page refreshs every 15min-20min (like pressing F5). 

Is that how it should be?

Regards
Maurice

MichMich, 2015-11-05:
Hi Maurice,

The full page refresh is not intended. This is probably a bug which I can't seem to replicate. Is there a way you can give the new code a try on your desktop and investigate if you see any errors in the console?

Cheers,
Michael

mattreese808, 2016-01-12:
Having same issue and the pi can't seem to refresh the page after the ~15 min interval, it just loads forever.  The only way is to click the url and press enter... which doesn't work in fullscreen and defeats the purpose of the autostart.

Is there anyway to disable this auto refresh?

nhubbard, 2016-01-17:
I had this issue and could only fix this with by disabling the version component. I deploy mine on an Azure site, so I can't even use the Git refresh feature.

bitte-ein-bit, 2016-01-28:
I have this issue to on my dev System which does not have a commandline git available for php. It's the version control for sure.

MichMich, 2016-01-28:
The auto reloading isn't working properly for me either. Feel free to remove or replace this feature in a pull request.

bitte-ein-bit, 2016-01-29:
@MichMich  are you directly pushing to the repro the pi is running from or does the pi have a seperate clone to serve from?

The first might be troublesome because HEAD changes way more often.

MichMich, 2016-02-06:
Maybe we could change this feature by allowing to refresh the screen by a command line command. Usually after a git pull, you still need to change the config file. Maybe we could do something like

    touch refresh

And let the mirror check the edit date of `refresh`

Any thoughts?

DanWilkerson, 2016-02-14:
@MichMich First of all, thanks for this project! I've built a full-length version (The glass is 59 1/2" x 20", and the whole assembly is 70" x 32") after being inspired by your original creation.

Per live-reloading, I do feel that moving away from the githash solution would be a good idea, since I often want to test things without committing new code, but that just might be me. I think having the front end poll the backend for an update indicator is a good idea. Not sure the best way to generate that at the moment, though.

MichMich, 2016-02-16:
@DanWilkerson , I agree. Those are cases when you want manual reloading. With a bit of luck I might be able to spend some time on this coming weeks, so hopefully I can address this issue.

PR #88 (2016-02-16): Switched to $.getJSON for auto-parsing to fix versions reloading
Fixes #35 (at least mechanically, doesn't address larger questions about how to handle live reloading).

$.ajax() returns data in a string, which would always return undefined when asking for data.gitHash. Switching to $.getJSON() corrects this.
closed 2016-02-16 by MichMich