MagicMirrorOrg/MagicMirror #36: Calendar not working for Google calendar

obrienk10, 2015-11-07:
It looks like the latest code no longer works with Google calendar dates.  It only shows recurring events now, not one-off events.  I'm no javascsript expert, but I took a look, and in calendar.js the 'seconds' variable always seems to be negative.  The obvious culprit is the piece of code marked "//FIXME: problems with Gmail's TimeZone", so I guess it's a known issue.

tilomitra, 2015-12-24:
It is for me. Follow the steps under "See your calendar" here: https://support.google.com/calendar/answer/37648?hl=en

obrienk10, 2015-12-24:
It used to work fine until I updated. The comment in the code even seems to indicate that it's known to be broken. Haven't tried the latest updates but will try again next week.

kichilron, 2016-01-08:
Seems to be doing the same thing for me - at the moment it's simply outputting a "500 server error" when I put in my ICAL-URL from Google.

tottidergeier, 2016-01-09:
i tried the "see your calendar" and even some easy basic .ics url, but i don´t get to see a calendar.
No error, just nothing.
Did i miss something? 
I just added my own URL into config.js

scooper86, 2016-01-10:
Yes, same thing for me since latest update.

Updated config.js as per and also tried entering it straight into the PHP file however no luck.

Nexxic, 2016-01-10:
Same error here. Have tried putting the .ics url both in the config.js and controllers/calendar.php.

I get the following message in the chrome debugging tool: 
ical_parser.js:32 GET http://192.168.1.9/controllers/calendar.php?url=https%3A%2F%2Fcalendar.google.com%2Fcalendar%2Fical%2my-private-calendar-url/basic.ics 500 (Internal Server Error)

kichilron, 2016-01-10:
That is exactly the error I'm getting.

danedb, 2016-01-10:
I had the same problem.
There's a parenthesis missing at line 33 in controllers/functions/gzip.php

```javascript
        if(stristr($h, "content-encoding") and stristr($h, "gzip") {
```
should be 
```javascript
        if(stristr($h, "content-encoding") and stristr($h, "gzip")) {
```

that change got it working for me...

Nexxic, 2016-01-10:
That fixed it for me too. Thanks!

kichilron, 2016-01-10:
Worked perfectly, thanks for investigating.

nhubbard, 2016-01-11:
I'm the person that made the server part modular, but I will submit a pull request to fix it. Thanks for letting us know!

tottidergeier, 2016-01-12:
Worked for me , Thanks so much!

xcellcior, 2016-01-14:
I tried this fix and still it doesnt work. Here's what I did from the beginning:

1. Downloaded the magic mirror zip from this site

JS>CONFIG.JS
2. added 'en' language on lines 2 and 13, 'Toronto,Canada' on line 10, and my API code for weather on line 14
3. removed the url from the news code on line 41 so it doesn't load news
4. swapped in my private ical google calendar url

CONTROLLERS>CALENDAR.PHP
5. swapped in my private ical google calendar url

CONTROLLERS>FUNCTIONS>GZIP.PHP
6. added the extra parenthesis on line 33

INDEX>PHP
7. removed <div class="news medium"></div> from line 20

Everything works except my calendar still wont load. I have the code on:
http://www.magicmirror.xcellwebdesign.com

Any help is much appreciated :)

jimmyly89, 2016-01-14:
Adding the bracket worked for me as well! Been trying to figure this out all week!

Thanks!

nhubbard, 2016-01-17:
This issue has been fixed as of acceptance of commit 91a57fdd88d331759462b8c312cd62d797fe9924. Please close this issue, @MichMich.

h-ramp, 2016-01-31:
Hi, I'm having the same issue as xcellcior. The extra parenthesis is added on line 33 and I'm using my gcal ical private address in config.js ... but no luck with the calendar showing. Any help is much appreciated.

closed 2016-01-31 by MichMich