MagicMirrorOrg/MagicMirror #37: Calendar event ordering

Stifler6996, 2015-11-10:
Hi,
I see the change that you placed in your code for max calendar entries. However the calendar doesn't always come back in the date order.

I fixed this by changing your code to ensure the sort occurs first. Then simply slice the final array.

Remove your new for loop, then

add this right after your array.sort
this.eventList = this.eventList.slice(0, calendar.maximumEntries);

This way it ensures all entries are sorted, and then the array is just culled to length.

Good work!
:)

MichMich, 2015-11-10:
Thanks for pointing that out. Could you make a pull request? 

Thanks!

mariners100, 2015-11-20:
Is there any update on this?  My calendar is still not sorting correctly.

Thanks!

Stifler6996, 2015-11-20:
I cannot make a pull request. My version is a fork from another fork. MichMich will have to make the change.

Attachment is what I would do.

Regards
[calendar.txt](https://github.com/MichMich/MagicMirror/files/39710/calendar.txt)

MichMich, 2015-11-20:
I just pushed an update. Slightly different than Stifler6996 proposed, but it should work. Please let me know it this solves the issue.

closed 2015-11-20 by MichMich