MagicMirrorOrg/MagicMirror #39: config.lang getting overwritten

fabiangigler, 2015-11-22:
Hey, awesome project!
Started looking into it and adapting it to my needs, I just noticed that updating the lang parameter in the config.js file does nothing for me, and still return nl in the browser console.
I couldnt figure it out myself by reading through the code, either theres a different parameter to change the language of moment.js or its a bug.
greetings,
fabian

Dom1n1c, 2015-11-29:
well, sometimes you need to clear cookies/browser history, because js files are cached.
i disabled the caching at config.js and main.js with the following command in index.php

    <script src="js/config.js?nocache=<?php echo md5(microtime()) ?>"></script>
    <script src="js/rrule.js"></script>
    //more code in here
    <script src="js/main.js?nocache=<?php echo md5(microtime()) ?>"></script>

xcellcior, 2016-01-14:
1. I added the parenthesis to the gzip.php file as stated above
2.  I added my private basics.ics url to the config.js in the js folder (also this js needs a semi-colon at the end says my dreamweaver?)
3. I added my private basics.ics url to the calendar.php in controllers folder

Still I can't see my google calendar. I hosted these files on one of my subdomains

magicmirror.xcellwebdesign.com

Can anyone see what I may have done wrong? I had no trouble with changing the weather to my locale, moving the random text to the bottom of the screen and removing the news.

Thanks!

bitte-ein-bit, 2016-01-31:
@xcellcior http://www.magicmirror.xcellwebdesign.com/controllers/calendar.php?url=https%3A%2F%2Fp01-calendarws.icloud.com%2Fca%2Fsubscribe%2F1%2Fn6x7Farxpt7m9S8bHg1TGArSj7J6kanm_2KEoJPL5YIAk3y70FpRo4GyWwO-6QfHSY5mXtHcRGVxYZUf7U3HPDOTG5x0qYnno1Zr_VuKH2M does not return anything. Maybe you want to pull the lastest changes. And check that all required files are there. Check in the apache error log for the website. I think php just might bail out with an error.

Also config.js does not seem to be updated on www.magicmirror.xcellwebdesign.com. At least it still says nl, etc.

And nexttime I suggest you open your own issue as I can not see any relation to the original issue.

MichMich, 2016-02-06:
Hi Guys, can I close this issue?

closed 2016-03-30 by MichMich