MagicMirrorOrg/MagicMirror #54: Few questions about PHP code

efiku, 2016-01-20:
Hello @MichMich 
Project MagicMirror is awesome. :+1: !
I've few questions about PHP code here 

1. Why you decide use include for importing css? 
```php  
    <?php include('css/main.css') ?> 
```
 In my opinion better option:
```html
 file: https://github.com/MichMich/MagicMirror/blob/master/index.php#L5

MichMich, 2016-01-20:
I initially used this to prevent caching. I agree that this currently isn't the cleanest way.

efiku, 2016-01-20:
Ok Issue edited. Please look now :smile: 
@MichMich I can make PR with some changes for PHP code if you allow of course.

MichMich, 2016-01-20:
Make sure you throughly test everything on a raspberry Pi. I think php 5 should be supported. (Not only 7).

MichMich, 2016-02-06:
Can I close this issue? Or do you need more input?

efiku, 2016-02-06:
I have no free time right now :(

closed 2016-02-06 by efiku

MichMich, 2016-02-06:
Same problem here. ;)