MagicMirrorOrg/MagicMirror #60: Netatmo Integration

CFenner, 2016-01-27:
I'm working on a Netatmo implementation for my mirror (which is by now just a web page) to display humidity, temperature, co2 of my different indoor stations and weather warnings. 

If anyone is interested in this I can do a pull request.

djsunrise19, 2016-01-27:
sounds pretty cool :).

paviro, 2016-01-27:
Why not write it as a module? :)
https://github.com/MichMich/MagicMirror/tree/master/modules

CFenner, 2016-01-28:
Thanks, I have'nt seen that. Are the modules loaded somehow automaticly?

paviro, 2016-01-28:
Yes they are :)
I changed the loader a bit further (not in the master yet) see https://github.com/MichMich/MagicMirror/pull/62/files to get a feeling of how to use them :) good luck if you have any ideas to improve the way modules are handled I am looking forward to hear them!

MichMich, 2016-01-28:
I merged that commit. - Not sure of the current module system is completely the way it should be. But is is a perfect first step.

paviro, 2016-01-28:
No it is surely not. I have never really written any PHP code, so if anyone can improve it and the way it handles anything I'd be really happy. I just didn't want to change the whole code every time there is an update to the master code so I had to start something :)

closed 2016-02-06 by MichMich

CFenner, 2016-03-22:
The Netatmo module is ready for the world ;) 

I finally had the time to put my code into the model structure. By now it can:
- display different weather modules of one Netatmo station
- display different sensor data of each module
- set the module order
- set the data order

What is missing: 
- time triggered data refresh (sorry, F5 only, may be integrated by the end of the week)
- formatting of the measured values

https://github.com/CFenner/MagicMirror-Netatmo-Module