MagicMirrorOrg/MagicMirror #63: Calendar URLs not changeable with js anymore

paviro, 2016-01-27:
I have a module changing the calendar URLs based on who is using the mirror. Before the multi-calendar update this used to work but now it does not any ideas?

`config.displaySymbol = true`

`config.calendar.urls = [
		{
			//Schule
			symbol: 'calendar-plus-o', 
			url: 'sdfsdf'
		},
		{
			symbol: 'mars',
			url: "werwer",
		}
		]`
		`calendar.updateData()`

MichMich, 2016-01-28:
Check with @bitte-ein-bit, he made the changes to the calendar.

bitte-ein-bit, 2016-01-28:
Putting the code below in $().ready function, has the desired effect. 

```
setTimeout(function() {
		calendar.displaySymbol = false;
		calendar.eventList = [];
		config.calendar.urls = [
		{
			symbol: 'calendar-plus-o', // Fontawsome Symbol see http://fontawesome.io/cheatsheet/
			url: 'http://www.ifeiertage.de/by-.ics'
		},
		];
		calendar.updateData(calendar.updateCalendar.bind(calendar))
	}, 10000);
```

The eventlist is not cleared anymore, only the events of the current calendar are purged. I think for switch calendars depending on the person I would prefer a solution with an added e.g. displayfor property which would make another round-trip to the web unnecessary. I'm not yet in this stage. so maybe I'll go down that road.

> config.displaySymbol = true 

I think you mean `config.calendar.displaySymbol` but anyway, yes this is just sourced once just like `config.calendar.maxEntries`.

Hope that helps.

closed 2016-02-06 by MichMich