MagicMirrorOrg/MagicMirror #66: News module not working

jerbookins, 2016-01-30:
I just pulled the new version today and uploaded it and the news module isn't working

I just changed the rss feed to "http://www.reddit.com/r/news/.rss", as I've done with many previous versions and not changed any internal settings.

> "No feed results for: https://query.yahooapis.com/v1/public/yql?format=json&q=select%20*%20from%20rss%20where%20url%3D'http%3A%2F%2Fwww.reddit.com%2Fr%2Fnews%2F.rss'"
> Failed to show a news story 20 times, stopping any attempts

bitte-ein-bit, 2016-01-31:
I get an error when I click on your second link
 `{"error":{"lang":"en-US","description":"Query syntax error(s) [line 1:62 mismatched character ' ' expecting ''']"}}`

You're sure the url is correct? Maybe reddit has some issues in their rss feed? Using a different feed it works for me.

MichMich, 2016-02-06:
@jerbookins Where you able to solve this? Can I close this issue?

linugen, 2016-02-14:
I am getting the same error as @jerbookins after replacing the feed with the AP feed    
```
news: {
        feed: 'http://hosted2.ap.org/atom/APDEFAULT/cae69a7523db45408eeb2b3a98c0c9c5'
    }
```
`{"error":{"lang":"en-US","description":"Query syntax error(s) [line 1:98 mismatched character ' ' expecting ''']"}}`

MichMich, 2016-02-14:
It's due to the usage of the yahoo api. In a previous version we did not use it. I'm not very happy with the Yahoo solution, but don't have the time to make a new solution for it. A pull request would be kindly appreciated.

MichMich, 2016-03-30:
This will be solved in V2.

closed 2016-03-30 by MichMich