MagicMirrorOrg/MagicMirror #68: simple questions about the code and possible changes

The-Flix, 2016-01-30:
First of all, thanks for the great project! I really like it and am trying to build my own mirror now. Before I start with the hardware project, I'd like to complete the website as best as possible for me. 

Right now I am thinking about using a touch panel in front of the mirror - I don't know if it'd look good or if it'd work at all. It just an idea at the moment. 

For that, I'd like to change the news display. I'd like open a small pop up window on the page when I click on it. This should display more about the currently displayed news headline and maybe have the possibility to load the complete article. And it should close if clicked next to the window on the background. 

Since I have no experience with php & javascript so far, do you have a way to realize it? Maybe a hint or an example so I could experiment with it myself? 

Cheers
Felix

bitte-ein-bit, 2016-01-31:
Felix, do you have any programming experience? Have you worked with HTML before?

You could add an extra [div](http://www.w3schools.com/tags/tag_div.asp) to display the extra news. No need for a popup here. The div can be used as an overlay to display the news content. Google and [Chromes Dev Tools](https://developers.google.com/web/tools/chrome-devtools/?hl=en) (F12) is a friend here when you want to find out what information is already available.

If you have no programming experience I would start out with a static html page which looks the way you want before you start solving the problem of getting dynamic data in there. 

Some phrases to google for would be [div overlay jquery](http://lmgtfy.com/?q=div+overlay+jquery),  [div overlay popup](http://lmgtfy.com/?q=div+overlay+popup)

Do you want to add a touch screen behind the mirror glass or do you want to add a touch foil between the mirror glass and the screen?

The-Flix, 2016-02-01:
Yes, I have programming experience. Just never had to use php & javascript. I will have a look at the google searches you posted. Thanks. 

The idea is to have the layers like Screen -> Mirror -> Touch. I have an old 22" screen which would be the base. On top a mirror and as most top layer, a touch panel. I found 22" resistive touch panels which should do just fine I guess.

MichMich, 2016-02-06:
Hi @The-Flix, thanks for participating in this project. Did you check if these touch panels work if you put glass (a mirror) in for of it?

The-Flix, 2016-02-06:
@MichMich, it won't work if I'd put the mirror in front of it. But I believe it should work if I put the touch panel on top of the mirror. It is not the best option because it is a resistive touch panel without multi-touch, but it should do the job just fine. 

Have a look at this: http://www.ebay.de/itm/22-Inch-5-Wire-Resistive-Touch-Screen-Panel-Kit-With-USB-for-22-16-9-LCD-Screen-/271027967970

I haven't bought anything. I will check if I get the website working the way I imagine and after that, I start looking for the hardware I could use. I have a unused 22" lcd screen which will display the website. A mirror, a frame, a touch panel, a PI will be bought later. :-) 

Unfortunately I have not much time to have a look at css, javascript, and php, but I will continue the work slowly.

closed 2016-03-30 by MichMich