MagicMirrorOrg/MagicMirror #74: News.js doubling some news feeds

miket82, 2016-02-02:
Hello. Thank you for the project and the people working on programming. 

I am new to the programming languages involved here. I have noticed that some RSS feeds (all of the Fark.com feeds) I tried display text twice in a row. It looks as if the RSS feed is valid if this website and Firefox can be trusted. I will attempt to go through the news.js code line by line and learn how it works and possibly how to fix it. Any insight on the issue would be helpful.

//Feed
feed: 'http://www.fark.com/geek/fark.rss'

//RSS Validator
http://feedvalidator.org/check.cgi?url=http%3A%2F%2Fwww.fark.com%2Fgeek%2Ffark.rss

MichMich, 2016-02-06:
I think this is an issue with the yahoo parser. I'm not very fond of this external solution. This solution might change in the future. Or feel free to send a push request.

Let me know when I can close this issue.

miket82, 2016-02-06:
I agree. I like your original solution much better and will try to make that work. As far as I am concerned you can close this issue. I am not skilled enough to isolate the problem to prove it is yahoo's problem.

MichMich, 2016-03-30:
This will be solved in V2.

closed 2016-03-30 by MichMich