MagicMirrorOrg/MagicMirror #76: Voice recognition

jamesarm97, 2016-02-03:
It would be nice if the efforts on this other fork can be worked in. I like the voice recognition portion.

https://github.com/evancohen/smart-mirror

paviro, 2016-02-03:
https://www.talater.com/annyang/ seems quite easy to add. If if find some time I might add it as a module :)

jamesarm97, 2016-02-03:
That is what I thought, especially since he started his mirror from this version anyways. It looked simple to implement the commands and response code. The Philip Hue code looks useful also.

paviro, 2016-02-03:
Could you start something? Have a look [here](https://github.com/MichMich/MagicMirror/tree/master/modules) for module support (will change in the future but should be easy to adapt to future changes). I am interested in helping since the feature seems nice but I don't have time right know because I first want to finish my own modules.

jamesarm97, 2016-02-03:
I will give it a shot. I just downloaded the code yesterday to play with since I resurrected a dead wide screen lcd ;) Looks easy enough to merge in. I will study the module format first to see what it takes to get up to speed. I ran the other code on my mac and really like the "Show map of xx" and how it looks on the mirror.

paviro, 2016-02-03:
It is probably a good idea to also have a look at this [pull request](https://github.com/MichMich/MagicMirror/pull/72) :)

bitte-ein-bit, 2016-02-04:
When implementing keep in mind that all audio is send out to some google server as long as annyang is active... so maybe add an stop listening command.

And by the way Evan Cohen's smart mirror is a completely different implementation. It was inspired by this project but they do not yet share code.

paviro, 2016-02-05:
Does it send all the time or only when someone is speaking above a certain threshold? We could at an optional Button via GPIO or something :)

gefangenimnetz, 2016-02-05:
Since you will probably only have limited vocabulary needed you could also consider using a local/offline voice recorgnition solution like Spinx (a JavaScript port is available).

paviro, 2016-02-05:
I used that a while ago.. Wasn't really usable for anything else than english without creating your own model or something might be a bit too complex for the average user (see [here](https://github.com/MichMich/MagicMirror/pull/59#issuecomment-176126655))

gefangenimnetz, 2016-02-05:
You are right @paviro one has to manually do all the tedious modeling stuff, splitting words into phonemes etc. But for a limited vocabulary this should not be to much work + extension developers could do this them self if they want their module to be voice activated.

If one would use Googles Speech recognition on the other hand it’s much easier, but that would result in a constant audio stream submitted to Google (actually only 60 seconds and then you have to restart as Google has this limit in place). Since a lot of people are not living alone that might be something to consider.

I can also imagine a mix of both worlds:
Spinx constantly listens (offline) for an activation phrase, like "Listen up mirror".
This could then activate Annyang (wich is using Google Speech Rec.)

MichMich, 2016-02-06:
Indeed, this would be a nice addition when we have the modules all up and running.

paviro, 2016-03-11:
I startet to code something http://metype.org/en/2016/03/sprachsteuerung/ :)

MichMich, 2016-04-05:
Since this now is an external Module, I close this issue. Feel free to reopen if necessary.

closed 2016-04-05 by MichMich

Vaggan, 2016-04-05:
I have started a voice recognition module for the old module system.

I currently only works with th sonos api. But it´s a start.

Find it [here](https://github.com/Vaggan/MagicMirror-VoiceRecognitionModule)