MagicMirrorOrg/MagicMirror #77: Printing calendar date as well as day until

phoenixrip, 2016-02-04:
Thank you so much for writing and releasing this beautiful interface :)
I'm just trying and failing to make a small adjustment to the calendar display.
I would love it to have the actual date before the event, but I just cannot for the life of my work out how to get the calendar.js to give me the event date.

Essentially what I'd love is this:

23rd June   Event Title    In 3 weeks

I've tried e.DTSTART and e.date and a load of other things but nothing will give me the start date :)
Any help would be massively appreciated.

bitte-ein-bit, 2016-02-04:
The events are added to a list in e.g. [calendar.js Line 115](https://github.com/MichMich/MagicMirror/blob/master/js/calendar/calendar.js#L115). This list is then worked with in the [updateCalendar](https://github.com/MichMich/MagicMirror/blob/master/js/calendar/calendar.js#L153) function.

You could use the seconds parameter to calculate the date or add the startDate as a parameter to all ```this.eventList.push``` calls.

[Chrome's DevTools](https://developer.chrome.com/devtools) are really helpful to look at js objects.

aramaki87, 2016-02-06:
@ Calendar.js
@about line 55: add
`var evDate = moment(e.DTSTART).format('L');`

@about line 66: add
`var evDate = moment(e.startDate).format('L');`

@about line 81: change to
`this.eventList.push({'eventDate':evDate,'description':e.SUMMARY,'seconds':seconds,'days':time_string,'url': url, symbol: this.calendarSymbol});`

@about line92: change to	`this.eventList.push({'eventDate':evDate,'description':e.SUMMARY,'seconds':seconds,'days':time_string,'url': url, symbol: this.calendarSymbol});`

@about line 110: add
`var evDate = moment(dt).format('L');`

@about line 121: change to
`this.eventList.push({'eventDate':evDate,'description':e.SUMMARY,'seconds':seconds,'days':time_string,'url': url, symbol: this.calendarSymbol});`

@about line 175: add
`row.append($('<td/>').html(e.eventDate).addClass('days dimmed'));`

That should do it...
Yo can even play around with the position if you move the transaction at 175 up or down.
Cheers.

phoenixrip, 2016-02-06:
:) this was the perfect answer! I was very close already but this was exactly what I needed! :)
I've actually used the moment .format(MMM Do) in the variables which gives me a lovely

Feb 7th - Event title - in 1 day

Thank you both so much for your help!

closed 2016-02-06 by MichMich