MagicMirrorOrg/MagicMirror #78: Weather Forecast - Low Temp Displayed in Wrong units

dpcarlton, 2016-02-05:
I am using imperial units in my config.js. The current weather and high forecast shows up in Farenheit, but the low appears to be in Celsius.

[image]

theanimator, 2016-02-05:
I'm also using Imperial, but mine appears to be correct even though there is quite a spread between the high and the low. I compared with http://openweathermap.org/city and weather.com

dpcarlton, 2016-02-05:
Anyway you can post a shot of your weather widget? My temps are not matching up at all.

theanimator, 2016-02-05:
I'm in Santa Clara CA [image]

dpcarlton, 2016-02-05:
Yeah, yours looks right. Out of curiosity, are you using lat/long or city name in config. js?

"            q: 'Colorado Springs,US',"

theanimator, 2016-02-05:
q: 'Santa Clara,US',

dpcarlton, 2016-02-05:
Could be an issue with my city. I changed to Santa Clara and it came up fine. Odd.

axlbundy, 2016-02-06:
I'm having the same issues with los angeles weather. I've tried Los Angeles,US and Los Angeles,California both give the same incorrect High/Low Temps. The really strange thing is that the current temperature seems to be spot on

MichMich, 2016-02-06:
If you use the openweathermap api you could check to see what the api feed sends to your mirror. If it is indeed an issue with the API, I would suggest reporting it to openweathermap.

Please let me know when I can close this issue.

dpcarlton, 2016-02-06:
Thanks. Is there a way to see the raw api data that I am receiving?

MichMich, 2016-02-06:
You could use the browsers web inspector, or just surf to the url:

    http://api.openweathermap.org/data//2.5/weather?q=Baarn%2CNetherlands&units=metric&lang=nl&APPID=YOUR_API_KEY_HERE&cnt=5

Make sure to modify the parameters in the link above.

dpcarlton, 2016-02-06:
Thanks, that url returns the current weather, are you using this for the
forecast?

This url seems to return the correct forecast.

http://api.openweathermap.org/data/2.5/forecast?id=5417598&units=imperial&appid=
<apiID>&cnt=5

MichMich, 2016-02-06:
you're right. sorry.... posted the wrong link.

dpcarlton, 2016-02-06:
Ok, so I see something.

In your code, you are using this url:

http://api.openweathermap.org/data/2.5/forecast/daily?id=5417598&units=imperial&appid=
<appid>&cnt=5

{"day":63.41,"min":31.37,"max":70.56,"night":41.77,"eve":62.76,"morn":31.37

This url returns completely different results then this url:

http://api.openweathermap.org/data/2.5/forecast?id=5417598&units=imperial&appid=
<appid>&cnt=5

{"temp":57.13,"temp_min":41.58,"temp_max":57.13,

dpcarlton, 2016-02-07:
So it appears some odd rounding is happening in the code. 

Here is the displayed data:

[image]

And here is the raw data:

[image]

Sun - Min is off by 3 degrees.
Mon - High is off by 2 degrees
Tue - High is off by 2 degrees.
Wed - High is off by 2 degrees.
Thu - Both Match
Fri - Both Match

bitte-ein-bit, 2016-02-08:
@dpcarlton Sorry to say, but no... this is just rounding to one decimal place. 37.13 becomes 37.1 and 21.49 is rounded to 21.5.
[image]

dpcarlton, 2016-02-08:
You are right, sorry. When I looked at it yesterday I could have sworn it was off. No idea how I mis-read it. You can close this issue. I opened one with the api providers.

closed 2016-02-09 by MichMich

jamesarm97, 2016-02-09:
FYI, looks like you were looking at the "current" temps vs the Max temps and that was the reason for the difference.