MagicMirrorOrg/MagicMirror #80: New Module System Discussion - Please contribute.

MichMich, 2016-02-07:
Maybe it's a good idea to discuss the new [Module System](https://github.com/MichMich/MagicMirror/wiki/Module-System-Proposal) before we writeout the API. 

@paviro, I noticed you've contributed. Thanks for this. I really appreciate your help! Just a few questions though: 

The functions you proposed, are those method of a module class? I can't really wrap my head around how this would work, could you elaborate?

Also, I think the added regions are a bit unnecessary. In most cases you can just add content in one of the corners (the content will stack) or align right or left in the upper third, center or lower third. I think we should try to keep the layout simple.

Aslo, what do you guys think about using a system like vue.js? I'm not a big fan of jQuery, and prefer a mvvm model. But this might make it to difficult for most users to contribute.

paviro, 2016-02-07:
To be honest with you I haven't thought all that much about how to implement it. I just added what I would need to port my modules to the new system, which is:

- Helper scripts written in Python and Node.js running as background daemons
- A way to switch between multiple views (one for general users and one for people who are logged in via facial recognition). I would therefore love the possibility to be able to define more than one module layout in config.js and also to switch them via a module. Might also be useful if you want to have different views for lets say work related and private stuff. A module could again provide an interface to switch between them for example via a GPIO button.

Since my edits in the wiki are not that useful in their current state, I will remove them, they are hear now anyway.

bitte-ein-bit, 2016-02-08:
This is going to be some random thoughts.

I think if you switch to vue.js most user will not be able to contribute. I'm not sure about stability. I once used a rather new php framework (symfony 1.x) and later regretted it because their api change a LOT when going to 2.0. IMHO jQuery seems to have stabilized. I can't tell for vue.js or angular.js. Never used either. On the other hand not using spaghetti code is nice.

Personally I think it would be a good idea if a module could listen and trigger events. Some events could be defined (e.g. 'User changed', 'User logged out', 'Calendar event starting', 'GPIO x HIGH', ...) and some central instance could route them. This core should be very minimal and all that is needed to run the mirror. I think that everything else should be a module.

I agree with @paviro that some modules should be implemented without any use by themself like `socket.io`. This makes it necessary that modules can specify depends.

Regarding the helper scripts I think it would be nice to have e.g. a Makefile which will download and install the dependencies and setup an init.d or systemd script. Writing an daemon is rather advanced so IMHO the developer of that plugin should be able to find enough information to provide a Makefile. Other options would be some sort of Chef cookbook, Puppet Manifest, Bash Script, ...  which could also setup the system itself. A Makefile would probably be the easiest.

I think the regions will work fine. But we should find a way, that a user can override all region positions without interfering with the main code. Not all mirrors are the same. So they might need some way to adjust margins, etc. For some modules it would be nice to render depending on the screen orientation but that is just eye candy.

paviro, 2016-02-10:
I think I will write a Python CLI which could be used to install modules and also list all available modules with a description. I could implement the systemd or init.d scipt setup there :)

## CLI
Something like:
- magic-mirror list (show all plugins with short discription)
- magic-mirror info module (show long description, dependencies etc. for module)
- magic-mirror install module (installs module and dependencies)
- magic-mirror remove (removes module and dependencies, if installed via magic-mirror install)
- magic-mirror enable/disable module (?)
- magic-mirror update (Update module list and versions)
- magic-mirror upgrade (install updates)

MichMich, 2016-02-10:
Hi @paviro, before we start building command line tools, It might be a good idea to first finish of the basic concept behind the modules. How they work, and how dependencies are loaded. - Also, since most of it is javascript based, we might consider using existing CLI tools, like npm. This way, we can put our effort in the enhancement of the mirror UI.

MichMich, 2016-02-10:
@bitte-ein-bit I agree with the idea of event triggering. Besides a module instance, there should be indeed a MagicMirror shared instance (singleton concept) which can receive messages of the module instance (in other words, a delegate for the module). 

The MagicMirror share instance (the main instance) could have methods to load and unload modules. A broadcasting system for notifications between modules (so no direct module to module communication). A method to reload the HTML/dom for a specific module ... etc.

Regarding the overriding for region positions: this can all be done by css. The region is just a simple way to load a module in a specific location, but the module is free to adjust it's design using CSS. (As long as it doesn't interfere with other visuals.)

CSS based, it might be a good idea to define some standard elements modules can be used. If all modules use the same css elements, we could also implement a tempting system in later stage. 

Regarding dependencies and installing libs. I think this should be taken case of if we have the base of the module system running. IMHO we could use existing systems like NPM.

Most importantly: thanks for your effort and contributions. I love the fact we are working on the future of the Magic Mirror.

MichMich, 2016-02-10:
As a small side note: If we start using NPM, we could also consider building a small node web server to host the magic mirror. This way, users don't need to install Apache & PHP.

gefangenimnetz, 2016-02-10:
I would love a JS only implementation. That would allow for easy porting to to Android or IOS using build tools like cordova (phonegap), etc. Basically that’s what I am currently doing.

paviro, 2016-03-02:
@MichMich Your are right :)

MichMich, 2016-03-02:
Not sure yet, but I might have some spare time end of this month to work on this version.

sclausen, 2016-03-08:
@gefangenimnetz I'm currently working on this. It's based on angular2 and I'm still not sure, if I'll add a backend. Meteor (still with an angular2 frontend) is one idea and works really well, especially with realtime server side functionality like motion detection.

paviro, 2016-03-09:
@MichMich  @bitte-ein-bit  any news on your side? Any way I can help you with the module system with my limited JS knowledge?

sclausen, 2016-03-10:
@gefangenimnetz @MichMich I started a javascript-only clone [sclausen/MagicalMirror](https://github.com/sclausen/MagicalMirror)

bitte-ein-bit, 2016-03-11:
Unfortunately I'm packed with other stuff to do. Don't have the time to
lock into a new backend right now. Sorry.

sclausen, 2016-03-11:
@bitte-ein-bit currently it has no backend and works frontend only with calendar, news, weather and time.

MichMich, 2016-03-18:
I'll be probable be working on a new version coming weeks. I'll keep you guys posted.

paviro, 2016-03-18:
Nice looking forward to it!

MichMich, 2016-03-23:
Hi Guys,

Today, I FINALLY started working on the module system. Although I only have the very simple foundation up and running, I think I'm really on the right track.

A small overview so far:

**config.js**
Used to define all the modules that need to be loaded, including the configuration for the specific module.
Adding a module to this config will ensure the module will be loaded, including all nessecery .js and .css files. So other then downloading a module to the module folder and adding it to the config, no further actions are required to add it to the mirror.

```javascript
	var config = {
		modules: [
			{
				module: 'clock',
				position: 'middle_center',
				config: {
					foo: 'Bar'
				}
			},
			{
				module: 'compliments',
				position: 'middle_center',
				config: {
					compliment: 'You look hot!'
				}
			},
			{
				module: 'compliments',
				position: 'top_right',
			}
		]
	};
```
**modules/compliments/compliments.js**
This is a very simple example of a module. Of course, this doesn't include any logic yes, but shows how simple a module is. 

```javascript
	Module.create({

		// Define the default config. This is optional.

		defaults: {
			compliment: "This is the default compliment"
		},

		// Define a start method. For example if you want to schedule timers,
		// or want to make connections to socket servers. Again: this is optional.

		start: function() {
			Log.info('Starting module: ' + this.name);
		},

		// Return an array of scripts that need to be loaded.
		// All scripts are loaded before the module is started.
		// Loading might be skipped if the file is already loaded.
		// The system contains a vendor folder for default libs like jQuery.
		// See comments below for more info.
		// This method is optional.

		getScripts: function() {
			return [
				'dummy.js', // Loaded (if available) form vendor folder, otherwise from module folder. Loaded only once.
				'http://server.com/js/file.js', // External file. Always loaded.
				this.file('foo.js'), // File from module folder. Multiple foo.js files might be loaded.
			];
		},

		// Same system as getScripts().
		// This method is optional.

		getStyles: function() {
			return [
				'compliments.css'
			];
		},

		// Returns the dom to be inserted in the mirror.
		// This method is optional.

		getDom: function() {
			var compliment = document.createTextNode(this.config.compliment);       
			var div = document.createElement("div");
			div.appendChild(compliment);

			return div; 
		}
		
	});
```

And thats all there is to write a module! Of course, I will continue to work on this. If I have any news I'll let you know. Hope to share a basic version next week. 

All code so far is pure javascript. But of course, modules are free to use additional libs like jQuery.

Open for feedback.

CFenner, 2016-03-23:
Looks great. I like the js only approach.

MichMich, 2016-03-23:
And, as an example, a super simple module to show something on the mirror:

**modules/helloworld/helloworld.js:**

```javascript
Module.create({
	defaults: {
		text: "Hello World!"
	},
	getDom: function() {
		return document.createTextNode(this.config.text);  
	}
});
```

This allows you to display a text on the mirror via the config: 

```javascript
var config = {
	modules: [
		{
			module: 'helloworld',
			position: 'middle_center',
			config: {
				text: 'This works perfect!'
			}
		},
	]
};
```

paviro, 2016-03-24:
Looks great! :) How are we going about socket connections? Should each module create their own (also if it uses helper scripts) or should we create a connection module which serves information from one socket to all the modules which need it and helper scripts have to talk to our socket?

MichMich, 2016-03-24:
I've implemented a notification system to communicate between modules. This will also be used for the incoming socket messages, and probably also for the outgoing messages (although I don't see a use case for outgoing messages yet ...).

paviro, 2016-03-24:
There is a use case. I would like to implement voice control and also use that for example to set my alarm clock which would need outgoing connections.

MichMich, 2016-03-24:
But in that case you might want to communicate with a socket other then the mirror's backend. For now, the backend won't include a module system. (Will definitely be on the todo list.)

MichMich, 2016-03-24:
I pushed a very simple first version of the v2-beta to a separate branch. Unfortunately I did not have the time yet to write a readme, but most files have sufficient comments. I probably won't be able to work on this version coming few days, but will definitely continue the project next week.

Feel free to give it a spin!

https://github.com/MichMich/MagicMirror/tree/v2-beta

Miyasashi, 2016-03-25:
Can someone explain what the module system does I dont really get it.

MichMich, 2016-03-25:
It will (eventually) allow everyone to built modules for the magic mirror without the need of modifying the core of the system. This way everyone can provide modules that can be installer on the mirror.

Miyasashi, 2016-03-25:
Thanks for the fast response, funny coincidence i had problems with implenting my own stuff and I think this will help a lot.

paviro, 2016-03-25:
@MichMich Do you think we could move the whole project to [Electron](http://electron.atom.io/) once the module API is finished? Only the calendar part would have to be ported to JS right?

CFenner, 2016-03-25:
It's not so easy to get a more complex module to run. This variable structure differs and Q seems not to work.

MichMich, 2016-03-25:
@Paviro: Maybe in the future. It's not the current focus. 
@CFenner: I don't really understand what You mean. Could you elaborate?

CFenner, 2016-03-25:
As the module was accessible via the global variable it was a bit easier to chain up the methods with Q promises. Now I need to bind the correct module object to each handler method to be able to call the next method. But I think I'm making progress..

MichMich, 2016-03-25:
Hi @CFenner, I don't have any experience with 'Q promises'. Could you give an example what was possible in the old version, which isn't in the new version. Also note: the new version is in a very early state. So a lot of things might break in future versions, so please don't spent to much time on building stuff for the new version yet. :)

CFenner, 2016-03-25:
Before I had this: 
```
return Q.fcall(
	netatmo.request.token, netatmo.render.error
).then(
	netatmo.request.data, netatmo.render.error
).then(
	netatmo.render.all
).then(
	netatmo.update.wait
);
```

Now it looks like this:
```
var that = this;
return Q.fcall(
	this.request_token.bind(that), this.render_error.bind(that)
).then(
	this.request_data.bind(that), this.render_error.bind(that)
).then(
	this.render_all.bind(that)
).then(
	this.update_wait.bind(that)
);
```

But I think it is needed because Q is using the methods as a callback from the ajax request, but after all it's working now. 

[image]

MichMich, 2016-03-30:
@paviro the node_helper todo's are all gone. :)

What's more important, is that I integrated a socket system for the node helpers. Using it is pretty simple:

**In client side module:**
To send a notification:
```javascript
var notification = "JUST_A_STRING"
var payload = {anything: 'you want'}
this.sendSocketNotification(notification, payload);
```

To receive messages, make sure your client side module has the following method:
```javascript
socketNotificationReceived: function(notification, payload) {
	console.log(notification);
	console.log(payload);
},
```

**In node_helper.js:**
Create a socket object: 
```javascript
var MMSocket = require('../../js/socketclient.js');
var socket = new MMSocket('nameofthemodule');
```
To send a notification:
```javascript
var notification = "JUST_A_STRING"
var payload = {anything: 'you want'}
socket.sendNotification(notification, payload);
```
To receive messages, register a callback:
```javascript
socket.setNotificationCallback(function(notification, payload) {
	console.log(notification);
	console.log(payload);
});
```

**Note 1:** The connection will be established as soon as the client side sends it's first message. 
**Note 2:** All modules of the same type receive the same notifications. 
**Note 3:** For the client side modules, we're using subclassing to keep thing more simple. (No need to create a socket object for instance ...), I might work on a node_helper.js superclass as well in order to streamline the experience.

paviro, 2016-03-30:
Nice! I will later change my module to use it :)
The weather module looks really gorgeous!

MichMich, 2016-03-30:
Ok, slight change. I changed the node_helper setup to use a superclass. This way it works the same as the clients side modules, So to recap: 

To send a notification:
```javascript
var notification = "JUST_A_STRING"
var payload = {anything: 'you want'}
this.sendSocketNotification(notification, payload);
```

To receive messages, make sure your client side module or your node_helper has the following method:
```javascript
socketNotificationReceived: function(notification, payload) {
	console.log(notification);
	console.log(payload);
},
```

MichMich, 2016-03-30:
About that namespace: it doesn't create a lot of overhead, but never the less it will be fixed in the future. Currently all node helpers connect to the socket with their own connection. In the future they just use the overal socket connection. This will also fix the issue you mentioned.

This should not affect any of your code.

paviro, 2016-03-30:
Alright :+1:

paviro, 2016-03-30:
Do you think it would make sense to add some kind of notification UI module, so that modules can show alerts? I use [sweetalert](https://github.com/t4t5/sweetalert) for my callmonitor but maybe it would make sense to integrate something like that into the core system? I am thinking of slide-in notifications (OS X style) but also UI overlays like I use in the monitor.

MichMich, 2016-03-30:
Alerts: this is exactly what I was thinking. This would be a nice feature for the future.
Swap module: yes this is on the to do list.

CFenner, 2016-03-30:
An alerts system would be a nice idea so each module can display configuration or communication issues in a similar way. 

Maybe build a module for it and use the socket system.

MichMich, 2016-03-30:
There already is a inter-module notification system. So we just need to built this module.

To send a notification to all modules (client side, so no socket connection):
````javascript
sendNotification: function(notification, payload);
````

To receive notifications:
````javascript
notificationReceived: function(notification, payload, sender) {
		if (sender) {
			Log.log(this.name + ' received a module notification: ' + notification + ' from sender: ' + sender.name);
		} else {
			Log.log(this.name + ' received a system notification: ' + notification);
		}
	},
````

MichMich, 2016-03-30:
@paviro: That namespace issue is now solved. https://github.com/MichMich/MagicMirror/commit/b243d25399e9538a0dc563e3fd7dd58c864937c1

paviro, 2016-03-30:
Nice! Glad module swapping is on the todo list!

MichMich, 2016-03-30:
What would be the best approach in your oppinion?

My idea is to allow the configuration of additional module classes in the config.

````javascript
var config = {
	modules: [
		{
			module: 'calendar',
			position: 'top_left',
			classes: 'michmich'
		},
		{
			module: 'calendar',
			position: 'top_left',
			classes: 'paviro'
		}
	]
}
````
This way you could do stuff like: 

````javascript
// using the default classes
MM.hideModulesWithClass('calendar',  speed, callbackFunction);

//using the custom classes
MM.hideModulesWithClass('michmich',  speed, callbackFunction);
MM.showModulesWithClass('paviro',  speed, callbackFunction);
````

Any ideas?

paviro, 2016-03-30:
I like that way :)

MichMich, 2016-03-30:
Cool, will try to implement it asap.

paviro, 2016-03-30:
Nice, I will try to finish my facial recognition module later tonight.

MichMich, 2016-03-30:
Great idea. Will keep it in mind.

Calendar module is almost finished. Will push it tomorrow.

paviro, 2016-03-30:
Nice! Is there currently a standard way of reading a modules config from the node helper script? Other than by require it?

MichMich, 2016-03-30:
I thought about it, but since a module can be included more then once, it has no use to push the config of the client to the node_helper. None the less, you should be label to just use the `config` variable (i think it's available in for the helper as well since it's a global ...), best practice would be to send the configuration from a module using the sockets when you perform a socket request.

paviro, 2016-03-30:
Yeah but my face reg module should not be loaded more than once anyway would not make any sense. Does not seem like I can access it via `this.config` just returns `undefined`. Would be great if it would be possible, since the module will use a python script and I want to allow the user to set the settings via the `config.js`.

MichMich, 2016-03-30:
You could try just `config` (without this.).

I think the best approach would be:

modulename.js
````javascript
start: function() {
    sendNotification: function('CONFIG', this.config);
}
````

node_helper.js
````javascript
socketNotificationReceived: function(notification, payload) {
     if (notification === 'CONFIG') {
         this.config = payload;
         //start setup of the python object.
    }
}
````

paviro, 2016-03-30:
I can't figure out what I am doing wrong...
I used `sendNotification: function('CONFIG', this.config);` but than I got an error` Uncaught SyntaxError: Unexpected token (`. If I do `this.sendNotification('CONFIG', this.config);` nothing ever gets received in the node helper..

MichMich, 2016-03-30:
Could you post your code somewhere?

MichMich, 2016-03-30:
It looks ok. Can't really check it since I am on my phone. Did you pull the latest v2 version?

MichMich, 2016-03-30:
Also, why use the timer? You could start python when the config arrives. (IF it arrives. ;))

paviro, 2016-03-30:
I used the timer because I was scared that it could be started twice if someone else sends something as CONFIG...
I updated now, have you changed how the helpers are called? Because I now get `No helper found for module: MMM-FRITZ-Box-Callmonitor.`.

paviro, 2016-03-30:
Works now, updating did the trick, thanks!

MichMich, 2016-03-30:
So it's solved? 

Also, do something like:

If(!pythonStarted) {
   pythonStarted=true;
   pythonStart();
}

[54 more replies]

closed 2016-04-15 by MichMich