MagicMirrorOrg/MagicMirror #83: Translation calendar.shortRunningText

fabianhussl, 2016-02-08:
Hi,

first of all thank you for your great work! It is great fun to configure the mirror.

I have a problem with the translation of the calendar.
In config.js I set `lang: 'de'`.

When I have an all-day event (for example a birthday) at the actual day (now), I get
_first name last name hat Geburtstag	still 31 Minuten_

The _still_ should be something like _noch_ or _nach wie vor_.

Can someone tell me where the translation take place?
I went through `js/compliments/compliments.js`, but I have no experience so far.

Thanks,
Fabian

bitte-ein-bit, 2016-02-09:
@fabianhussl, this part unfortunally has no autotranslate yet. See https://github.com/MichMich/MagicMirror/blob/master/js/calendar/calendar.js#L15.

fabianhussl, 2016-02-09:
resolved #83
thank you @bitte-ein-bit then it's clear for me! so far no autotranslate - just `still` resp. `until` driectly written in the variables.

closed 2016-02-09 by MichMich