MagicMirrorOrg/MagicMirror #87: How long will it take me to get to work?

eyeinthesky35, 2016-02-14:
Putting together a mirror and figuring out what can be done with this. An idea I had was a small section returning the time it would take for me to get to work with traffic.

I did some research and tested Google Maps DistanceMatrix API, which I think would be perfect for this. Unfortunately, I have ZERO skills in JS or CSS, so I have the idea and the tool, but no way to show the result. 

Anyone interested in implementing this?

The API query is simple:

https://maps.googleapis.com/maps/api/distancematrix/json?origins=(Home Lat/Long)&destinations=(Work Lat/Long)&key=(Super Secret API Key)&departure_time=now 

This returns:

`{
   "destination_addresses" : [ "(Work address, not lat/long surprisingly" ],
   "origin_addresses" : [ "home address" ],
   "rows" : [
      {
         "elements" : [
            {
               "distance" : {
                  "text" : "28.0 km",
                  "value" : 28045
               },
               "duration" : {
                  "text" : "22 mins",
                  "value" : 1347
               },
               "duration_in_traffic" : {
                  "text" : "21 mins",
                  "value" : 1242
               },
               "status" : "OK"
            }
         ]
      }
   ],
   "status" : "OK"
}
`

The only important return is duration_in_traffic, since it's usually a set commute. 

Another idea to add would be a stoplight that would show a different color based on the difference between average duration vs traffic duration (if slower by 5-25%, yellow; if slower by >25%, red).

What do you all think?

MichMich, 2016-02-14:
This would be a nice module. We're still working on the modules system though ...

eyeinthesky35, 2016-02-14:
Is there anything a Javascript and css noob can help with? I'm experienced with Linux but not Web design.

As an aside I got this going on an old eeePC 701 I had laying around then integrated the motion application and a usb webcam to turn the monitor on and off based on motion detection. Looking into motion control now.

MichMich, 2016-02-14:
We really need to finish off the Javascript first. Check out this issue: https://github.com/MichMich/MagicMirror/issues/80

eyeinthesky35, 2016-02-15:
Would you like me to close this issue?

closed 2016-02-15 by MichMich

SamLewis0602, 2016-04-14:
Did this module ever pan out? If not, I may give it a shot.

MichMich, 2016-04-14:
Please do @SamLewis0602! Let me know if you run into any issues with the API.