MagicMirrorOrg/MagicMirror #90: Git ignore file...

Snille, 2016-02-18:
Hi, I love the clean look of the "mirror".
It would be great if you added a .gitignore file ignoring the changes made in the config.js file locally.
When pulling the config.js file get's overwritten. :)

Thank you for sharing!

wcdanh2, 2016-02-18:
You can also ignore the config.js file on your end by adding it to .git/info/exclude  This will allow you to use "git add ." while still skipping changes to the config.js  (This is great for making a pull request back to the main branch.)

I expect that config.js will change its form in the future so when making a commit it will probably be best to cherry-pick the rather than ignore it completely.

Snille, 2016-02-18:
Ah, thank you. I'll do that. good to know. :)

closed 2016-02-18 by Snille