MagicMirrorOrg/MagicMirror #91: Google Calendar not working

kyle-ayn, 2016-02-26:
Everything is working, except my google calendar. I put in my url in the config file but it's not showing up.
[image]

bitte-ein-bit, 2016-02-26:
You might want to try to comment out the empty second entry. If this
doesn't work, do you see any errors in the development console of chrome
(F12)?

kyle-ayn, 2016-02-26:
Thanks for the suggestion, but it didn't work.
[image]

I looked at the console on Google Chrome on my PC, but it looks like nothing is wrong related to the calendar.
[image]

It also doesn't show up in Midori on my Raspberry Pi.
[image]

wcdanh2, 2016-02-27:
It looks like it is a group calendar.  Are you the owner?
Group calendars cannot be downloaded using the standard method, you must use the google API with Oauth2.  Currently this is unsupported.

I have a fix that I have been testing and plan to make a pull request in the next few days.

kyle-ayn, 2016-02-27:
The address is a public calendar for the school I go to. I'll try testing it with my own private calendar.

closed 2016-02-27 by kyle-ayn