MagicMirrorOrg/MagicMirror #97: Error on years with FB birthday calendar

FurNocte, 2016-03-06:
I've tried to use my personal FaceBook account to get birthday calendar, and it work. Only dates are too late off one year (as you can see on the three last lines).
[image]

monsieurpouet, 2016-03-17:
Perhaps, you can reduce size of the calendar div no ?

SVendittelli, 2016-03-25:
I am experiencing the same issue, has anyone found a solution?

@monsieurpouet I'm not sure what you mean by that, can you elaborate?

MichMich, 2016-03-30:
Please let me know if I can close this issue.

delize, 2016-03-31:
@FurNocte / @SVendittelli 

I am more interested in to how you got them to appear. 

I can't seem to get webcal:// to work, importing the webcal link into my gmail to use Google's ical link doesn't work, etc.

Could you describe how you did that?

At that point, if I experience the same problem, I will let you know.

SVendittelli, 2016-03-31:
@delize You can just replace `webcal://` with `http://`. So in the config for calendar urls you might have:

`{
	symbol: 'birthday-cake',
	url: 'http://www.facebook.com/ical/b.php?uid=1234567890&key=ABCDEFGHIJKLM_XY',
},`

I think it is because facebook only gives a start date and duration in the iCal but I'm not sure. Example birthday event:

`BEGIN:VEVENT
DTSTART:20160404
SUMMARY:Some Person's birthday
RRULE:FREQ=YEARLY
DURATION:P1D
UID:b123456789@facebook.com
END:VEVENT`

MichMich, 2016-03-31:
This should be fixed in V2. (I just finished the calendar module.)

SVendittelli, 2016-04-04:
I believe the update (46312cb) fixing recurring events broke the Facebook birthdays again, but I can't work out why.

MichMich, 2016-04-04:
I'll look into it tomorrow. While we are on the subject of Facebook events. Does somebody know how to convert an events page of a Facebook brandpage to an iCal Calendar? For example this page: https://www.facebook.com/DashBerlinOfficial1/events

SVendittelli, 2016-04-04:
Yes, select one of the events on the page and click going. There should be the option to export the event.
[image]
Then the ical link should be there!
[image]

MichMich, 2016-04-05:
Should be fixed again, now. If not, please reopen this issue.

closed 2016-04-05 by MichMich

MichMich, 2016-04-05:
For anyone looking for a nice solution: If you want a compleet feed of all events of a specific fan page, check out https://www.wallflux.com/#other

commit 598cf61e (2016-04-05): Fix Facebook Birthdays - Fix: #97