MagicMirrorOrg/MagicMirror #98: AM/PM

Chyase, 2016-03-08:
Is it possible to get AM/PM on the js?

Thanks,

monsieurpouet, 2016-03-17:
He is there.
In config.js, you've a parameter to change that. At the top of the file.

Chyase, 2016-03-17:
```
var config = {
    lang: 'nl',
    time: {
        timeFormat: 12,
        displaySeconds: true,
        digitFade: false,
    },
```

Where at?

danynek, 2016-03-17:
Time format: 12 or 24

DM

Chyase, 2016-03-17:
I'm actually looking for it to say if its AM or PM- not 24 hours vs 12 hours-

monsieurpouet, 2016-03-17:
Yes, I thought about this.
Maybe, I didn't understood the question. You talk about AM/PM and not 12/24.
For me, it was the same things.

Chyase, 2016-03-17:
Yeah I just want it to show AM when its morning and then PM when its afternoon/night

jerome2710, 2016-03-18:
You might use the format character `a` to show AM/PM.

Have a look at js/time/time.js near line 74:
`$(this.timeLocation).html(_now.format(this._timeFormat + ':mm[<span class="sec">]ss[</span>] a'));`
Note the added `a` near the end, this will result in AM/PM.

Good luck!

Chyase, 2016-03-18:
Jerome, that worked. Thanks!

closed 2016-03-18 by MichMich