MagicMirrorOrg/MagicMirror #102: MagicMirror is not formatted

DominicDecocco, 2016-03-23:
Hey guys, 
It seems that the Interface´s formatting is not active.
https://gyazo.com/c983b006ad854eb498de238a58bcb3fc

I guess it´s the same issue which is solved https://github.com/MichMich/MagicMirror/issues/89 but I don´t know what to do / What is the right directory of the CSS files?

MichMich, 2016-03-23:
Check your web inspector.

DominicDecocco, 2016-03-23:
In time.js line 26
`this.intervalId = setInterval(function () {`
it displays "anonymous function"

DominicDecocco, 2016-03-25:
In **time.js** in line 26 there is an _anonymous funktion_
`this.intervalId = setInterval(function () {`
but i dont think this causes my problem?!
Sorry I don´t have so much knwoledge about this.

MichMich, 2016-03-25:
Could You post a screenshot of your web inspector?

DominicDecocco, 2016-03-25:
[image]

MichMich, 2016-03-25:
Could you show what's in the console part?

DominicDecocco, 2016-03-25:
[image]

DominicDecocco, 2016-03-25:
[image]

bitte-ein-bit, 2016-03-25:
Are you using a web server? Looks like you are using a file:// link?

ski7777, 2016-03-29:
looks like php is not installed

closed 2016-03-30 by MichMich

DominicDecocco, 2016-04-02:
Sorry guys, I was on holiday..

I reinstalled everything again..copied the files to _/var/www/html/_
[image]

...after boot, chromium starts and it displays the same issue:(
[image]
[Webserver is running cause the console displays _localhost_;right?]

May I forgot something or made a mistake again?

MichMich, 2016-04-02:
What's the url Chromium uses?

DominicDecocco, 2016-04-03:
How can I check that?

MichMich, 2016-04-03:
What URL did you use in your autostart?

DominicDecocco, 2016-04-03:
http://localhost

MichMich, 2016-04-03:
If you click on the CSS folder in your debuggers left column. Doest it show a css file with contents? (In other words, does it load the CSS file?)

DominicDecocco, 2016-04-03:
It shows
font-awesome.css
weather-icons.css

MichMich, 2016-04-03:
It appears the main css file isn't loading. Is it in the css folder?

DominicDecocco, 2016-04-03:
Ok. Yes it is in the folder.

MichMich, 2016-04-03:
Sorry, I just remember the main CSS isn't included the usual way. It uses some PHP including: `<?php include('css/main.css') ?>`

That fact that this doesn't work probably is because your PHP isn't working. Like @ski7777 said before.

DominicDecocco, 2016-04-03:
Well ok. Apache & PHP-Support are installed (according to your instructions _Webserver_) and the _index.php_ is in the folder, too.
What else can I do?

MichMich, 2016-04-03:
You could go adventurous, and try the new V2 version I'm currently building. Build it's in an early stage so I can't give any support yet. Checkout the V2-beta branch.

DominicDecocco, 2016-04-03:
Yes that would be an interesting option..
Could it also be issued by the raspberry I use..it´s pi 3?

MichMich, 2016-04-03:
Yes, the installation is much more simple. I didn't yet test it on a Raspberry Pi though. But user @paviro did, so it should be working.

Keep in mind it's on heavy development at the moment, so you might need to update it every so often.

paviro, 2016-04-03:
Haven't run it on a Pi for quite some time now but it works you just have to install nodejs. Going to setup the new version on my Pi, as well today.

DominicDecocco, 2016-04-03:
Yeah I´m gonna try it when v2 is finished.
Do you have any idea why the php is not working?