MagicMirrorOrg/MagicMirror #104: Interface a little bit laggy

djsunrise19, 2016-03-24:
Hi,

do you have any performance tips for the interface? I made two mirrors (1x Raspi 2 Model B, 1x Raspberry 3, both with raspbian and the configuration on Michael's Website). Since the new Jessie Version the interface is very laggy, especially with the fade fx (rss, calendar, wheather, compliments). I have the newest updates. On Pi 3 it's a little bit better than on Pi 2.

My GPU Memory is 128MB. Do you guys have the same problems? What is your configuration? Any tips? Another browser, ... .?

Thank you, regards,
Maurice

MichMich, 2016-03-24:
Hi Maurice,

I'm currently working on a new version. This one issuing CSS transitions. No ETA yet.

djsunrise19, 2016-03-24:
Hi Michael,

thank you for answering. Nice to know! Are there any things I could do right now? More GPU Memory?

Regards

CFenner, 2016-03-24:
I use 256mb GPU mem without problems. Did you try to increase yours?

djsunrise19, 2016-03-24:
No, I don't,  I'll try it tonight. Thank you ;).

djsunrise19, 2016-03-28:
Sorry for the delay. I've tested the interface with 256MB GPU. It's a little bit better but not so smooth like with raspbian wheezy. So I think it isn't a hardware problem. Is it necessary to use jessie now and in near future? I don't know where exactly the problem is but since I've used jessie the whole website performs slower. I also installed the system a few times from scratch. Nothing better.

That's the reason why I asked this question ;).

MichMich, 2016-03-30:
You should ba able to run it on weezy. We are currently in the development of a new version. This should save your issue as well. Check out the v2 branch if you want more info. (Note: this version isn't ready for general use.)

closed 2016-03-30 by MichMich