MagicMirrorOrg/MagicMirror #105: V2: remote access not possible

CFenner, 2016-03-26:
The v2-beta is working good so far. Do simplify the configuration, I run the web server on my laptop and let the PI connect to die one. But since v2 the screen stays black. The html is loaded (static html is rendered) but the JS is not working.
If I open the index file directly on the laptop (file:///path/to/index.html) I get the following errors: 
[image]

If I connect with localhost (localhost/path/to/index.html) it's working. Any ideas how to fix this?

MichMich, 2016-03-26:
You should always use it using a webserver (a future version will include a node server). Using it directly using the file:// methode will never be supported. 

Since its in a very early stage I can't yet give any further support.

CFenner, 2016-03-26:
ok, I'll try to move the content to the PI's web server, that should solve the problem.

closed 2016-03-26 by CFenner