MagicMirrorOrg/MagicMirror #115: Calendars and recurring events

SVendittelli, 2016-03-31:
The calendar module only displays the first event for recurring events. So it only displays these for which the first event is in the future.

[image]
[image]

The relevant [.iCal](http://sharing.calendar.live.com/calendar/private/627e39d3-45d4-4763-b343-0b919de89dfa/a47b1221-7993-40bc-a844-5b42098bd9dc/cid-12ee71079603e521/calendar.ics).

MichMich, 2016-04-01:
This issue should be solved now. Could you give it a try?

SVendittelli, 2016-04-01:
Yep, works great!

commit 46312cb4 (2016-04-01): Fix #115 - Recurring events.
closed 2016-04-01 by SVendittelli