MagicMirrorOrg/MagicMirror #117: MagicMirror module show/hide animation

paviro, 2016-04-01:
If you hide a module and the one underneath it should stays visible it jumps (see [YouTube](https://www.youtube.com/watch?v=TwXuK0CHWTg)) to its new position. Would it be possible to animate/slide it to the new position?

MichMich, 2016-04-01:
I experimented a bit with this, but it's rather difficult. I will work on this in the future, but for now it hasn't got the highest priority. I've put it on the todo list in the README.

nhubbard, 2016-09-07:
If I'm correct, you could use CSS3 Transitions, correct?

MichMich, 2016-09-07:
@nhubbard: it already does

nhubbard, 2016-09-08:
In this case, when a module is deactivated, is it possible to apply a data tag (such as `data-module-deactivated=true`) to a tag and apply a transition/animation to that tag?

MichMich, 2017-01-19:
I don't think this is going anywhere soon. Fee free to reopen if necessary.

closed 2017-01-19 by MichMich