MagicMirrorOrg/MagicMirror #118: iCloud calendars not always working

paviro, 2016-04-01:
As you know my iCloud calendar does not work. I just created another test calendar which worked as expected but mine still does not.. Any way I can debug this?

MichMich, 2016-04-01:
First of: I'm working on the calendar integration now, in order to fix https://github.com/MichMich/MagicMirror/issues/115, so don't make any changes to the calendar module (else I won't be able to merge the pull request.)

Your issue is probably caused by not using the correct url, are you using a public url?

paviro, 2016-04-01:
Yes, I am using the correct URL, tried it with another calendar in exactly the same way and it worked there.

MichMich, 2016-04-01:
Can you share the non working URL?

paviro, 2016-04-01:
Well it's my private calendar, I could send it to you via e-mail?

MichMich, 2016-04-01:
That's ok: michael [at] xonaymedia.nl

MichMich, 2016-04-01:
@paviro This should be fixed now. Could you give it a try and close this issue when it's fixed? https://github.com/MichMich/MagicMirror/commit/46312cb44c08f1b1805a141d52af6863b0f5f098

closed 2016-04-01 by MichMich

paviro, 2016-04-01:
Sorry is fixed wasn't fast enough.