MagicMirrorOrg/MagicMirror #126: rss feed with ISO-8859-1 encoding is not displayed properly

estres, 2016-04-02:
rss feed with ISO-8859-1 encoding is not displayed properly

test feed:
http://www.spiegel.de/schlagzeilen/index.rss
header:
<?xml version="1.0" encoding="ISO-8859-1" standalone="yes"?>

MichMich, 2016-04-03:
I've just pushed an update. Could you please test to see if it works? **Please note:** you need to run `npm install` before running since it has a new dependency.

MichMich, 2016-04-05:
Closing this issue for now. Please reopen if the issue persists.

closed 2016-04-05 by MichMich

MichMich, 2016-04-05:
**Update:** The encoding is now a configurable option. Please refer to the module's README. The default value is `UTF-8`.

CFenner, 2016-04-05:
I'll try this evening.

estres, 2016-04-11:
found some time to test it. 
works great. thx
:+1: