MagicMirrorOrg/MagicMirror #127: CPU Usage spikes at 99% with nginx

delize, 2016-04-03:
Hey,

it seems that processor usage spikes to 99% for a single core, using nginx and the magic mirror. While this isn't an outright bug, processor power is very limited on the CPU usage on the raspberry pi itself. Is there any way that you can find ways to optimize the code so that CPU usage doesn't spike so high?

MichMich, 2016-04-03:
It's probably due to the JavaScript animation. Could you give V2 beta a try?

MichMich, 2016-04-05:
Closing this issue for now. Please reopen if the issue persists in V2.

closed 2016-04-05 by MichMich

delize, 2016-04-05:
Hey, 

Sorry, I meant to respond back earlier, but have been a little busy with work.

I would actually like to use V2, but am unable to get electron working from the directions provided on the readme.md file.

delize, 2016-04-05:
Oh, it looks like the instructions have since been updated when I tried. 

Sure, I will give this another shot tonight/when I get the chance.

MichMich, 2016-04-05:
Just to make sure, how do you start the app? By any chance using SSH?

delize, 2016-04-06:
So running this over SSH produced the following:

> pi@sec-mirror:~/Downloads/MagicMirror $ sudo ./install.sh
> Reading package lists... Done
> Building dependency tree
> Reading state information... Done
> build-essential is already the newest version.
> curl is already the newest version.
> unzip is already the newest version.
> wget is already the newest version.
> 0 upgraded, 0 newly installed, 0 to remove and 2 not upgraded.
> --2016-04-05 21:01:52--  https://nodejs.org/dist/latest/node-v5.10.0-linux-armv7l.tar.gz
> Resolving nodejs.org (nodejs.org)... 104.20.22.46, 104.20.23.46, 2400:cb00:2048:1::6814:162e, ...
> Connecting to nodejs.org (nodejs.org)|104.20.22.46|:443... connected.
> HTTP request sent, awaiting response... 404 Not Found
> 2016-04-05 21:01:52 ERROR 404: Not Found.
> 

When trying to run the manual version for node.js:
wget https://nodejs.org/dist/latest/node-v5.10.0-linux-arm64.tar.gz

The download link seems to be failing at trying to find the variables you set, if I manually specify the latest version of nodejs for the Raspberry Pi 3

https://nodejs.org/dist/v5.10.1/node-v5.10.1-linux-arm64.tar.xz
(looks like the file extension was changed to xz)

And after editing the script to take that URL I get the following after unpacking:

> node-v5.10.1-linux-arm64/share/doc/node/gdbinit
> 
> ## Testing Node.js and npm installation ...
> 
> + mkdir _test-node-install && cd _test-node-install
> 
> ## Creating test package ...
> 
> 
> ## Installing dependencies ...
> 
> + npm install --spin=false --loglevel=info --cache=.npm-test-cache
> /usr/local/bin/node: 17: /usr/local/bin/node:ELF▒Тs@▒▒@@%"@@@@@▒@@@▒▒▒▒▒▒▒▒Y▒▒Yx▒p▒▒▒▒Y▒▒Y0@@DDQ▒tdR▒td▒▒▒▒Y▒▒YXX/lib/ld-linux-aarch64.so.1GNUGNULD▒@▒▒▒e3▒▒+▒▒▒V▒
>                                                                                                                                                                   0▒▒▒@▒`#C
> 4@  B▒▒@`$@▒▒TB▒LhH"P▒
> "▒!▒A@(▒%H▒F▒0! ▒2▒T#(@&(@▒PH
> ▒▒▒0▒"E▒ @"▒&@3▒!▒+▒$▒HJA`▒ ▒ɀ(@@  M▒ꪔ▒▒XP1▒▒▒▒▒2F      ▒▒▒L1▒▒$▒▒▒▒Pl&▒▒$2▒$▒$d▒a▒˂▒▒X
> ▒!\&
>  Ȁ(▒A▒▒b▒)▒▒ AO@@S,5TT B"(C▒.▒H▒▒▒t     1BH▒▒▒▒"▒▒b@
>          L▒(▒▒x!▒▒p ▒"8@ ▒&▒▒▒▒▒$0
> ▒0▒2▒▒
>         0
> ▒a▒▒▒▒▒&▒@$▒5▒
> (g▒1▒(                                                                                                                                                                                              ▒▒
> @▒▒`G▒T▒▒ ▒`▒
> ▒0▒▒#%▒b▒$▒!▒2A▒
> ▒▒!     A*@L
> ▒k\$
> ▒▒ ▒(1M  ▒$▒ A▒PU;▒`▒▒
> *A▒L▒0F▒ v      ▒ A ( ▒]@@pf▒▒`▒▒dB
> ▒bG"▒@ ▒B▒▒d2L0a▒ ▒#▒▒lA@ ▒
> ▒PY*HkPÄ▒$▒B▒hN BB▒t8K▒d(▒@▒!DT ▒(|q▒@`▒E▒▒▒▒▒@▒D▒*▒H@I!▒@b     `bYdD▒▒*▒▒J▒▒
> C,▒▒0▒
> ▒▒Z@�pD▒PrQ"x▒▒T#A▒  b▒CP       ▒                                                                                                                                                                   ▒00      bHR▒▒▒▒Ԙ▒&▒j▒
> !▒*▒▒e▒▒A▒V@C! ▒!b@     %▒▒#B( bp▒X bP@@▒ ▒@▒Tu ▒▒&▒( L▒"GHb@ ▒▒        ▒▒*▒1▒ W▒@▒P @
> 4▒▒▒#(▒%*▒@▒▒)`d▒              D8▒CH ▒▒▒8(@|▒
> J▒4▒l4▒▒▒▒T▒▒ &▒▒I▒ք▒9   `▒ ▒9▒▒▒Q*uT▒H▒I`T▒▒▒+ <▒▒!▒ ▒HP8▒▒
> ▒$      ▒▒
> ▒▒`ʂ1@\`▒▒P▒wQT "▒z▒@ ,@Q&▒!@   @▒▒ ▒▒▒
> ▒$H▒,!▒p▒$@!B▒
> ▒b▒▒7͂F/,   H@▒E) !▒     ▒▒ !▒ Ȉ0DP
> 
> 
> F▒▒a▒   ▒▒▒▒▒▒8▒▒P
> R▒"DP@8D▒`▒▒▒łLD0 ▒ ▒(0▒IJ▒▒ ▒"▒@R▒▒▒   ▒e▒▒1▒""
> >▒l-8DJ▒@c▒,U▒LH$▒▒@h▒B▒: not found
> /usr/local/bin/node: 18: /usr/local/bin/node: Syntax error: ")" unexpected
> Error executing command
> 
> ## Cleaning up ...
> 
> + rm -rf _test-node-install
> PuTTYPuTTYPuTTYPuTTYpi@sec-mirror:~/Downloads/MagicMirror $ PuTTYPuTTYPuTTYPuTTYPuTTYPuTTYPuTTYPuTTYPuTTYPuTTYPuTTYPuTTYPuTTYPuTTYPuTTYPuTTYPuTTYPuTTY^C
> 

I then tried the same commands on a VNC session on the X console, and got the same result.