MagicMirrorOrg/MagicMirror #128: npm ERR

djsunrise19, 2016-04-03:
Hi,

I just wanted to test the new v2 beta and installed a new raspbian jessie, node 4.4. After that I copied the branch to my new folder /home/pi/mm. In that folder I type `npm install` at first. So far so good. After `npm start ` I get this: http://prntscr.com/anjz69

Did I forget anything?

Regards
Maurice

paviro, 2016-04-03:
Please send us the log shown in the last line!

djsunrise19, 2016-04-03:
Okay, here it is:
[npm-debug.txt](https://github.com/MichMich/MagicMirror/files/201301/npm-debug.txt)

paviro, 2016-04-03:
Thats weird can you try `npm install && npm run start` again?

djsunrise19, 2016-04-03:
Sure. Same result in console. Here the log file:
[npm-debug.txt](https://github.com/MichMich/MagicMirror/files/201302/npm-debug.txt)

CFenner, 2016-04-03:
Did you stop your Apache? I needed to change the port electron was running on. 80 didn't work as it was used by Apache, I changed the config to 5006 and that was ok.

CFenner, 2016-04-03:
Ah, you did a reinstall probably without Apache...

paviro, 2016-04-03:
Ah only root users can bind port 80! Try as root or change the port in the config.js

djsunrise19, 2016-04-03:
I changed the port in the config.js to 5006, same npm err in my console. Here the log file:
[npm-debug.txt](https://github.com/MichMich/MagicMirror/files/201303/npm-debug.txt)

paviro, 2016-04-03:
Does it work if you run it as root?
No you no longer need apache.

djsunrise19, 2016-04-03:
I check `npm install && npm start` again with the root user, still the same errors:
[npm-debug.txt](https://github.com/MichMich/MagicMirror/files/201313/npm-debug.txt)

paviro, 2016-04-03:
I'll try installing it myself after lunch and see if it works. Worked last time so know idea what's the problem.

djsunrise19, 2016-04-03:
Thank you! Enjoy your meal! :)

MadMathWizard, 2016-04-03:
I will add that I have the same issue.

paviro, 2016-04-03:
Alright guys...
1. Remove `node` and `npm` if you already installed via `apt-get`
2. Download the latest node version :
- `wget https://nodejs.org/dist/latest/node-v5.10.0-linux-armv7l.tar.gz` for Pi 2
- `wget https://nodejs.org/dist/latest/node-v5.10.0-linux-armv6l.tar.gz` for Pi 1
3. Unpack file `tar -xvf filename`
4. Install `cd foldername && sudo cp -R * /usr/local/`
5. Set loglevel `npm config set loglevel info`
6. `npm install && npm start` should work now :)

MichMich, 2016-04-03:
Great work @paviro! Could You add this information to the readme?

paviro, 2016-04-03:
Already did ended up in my pull request :/

MichMich, 2016-04-03:
An what about Pi3 is that the same as Pi2?

paviro, 2016-04-03:
Not sure, see [raspberrypi.org](https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=140572). Will check.

paviro, 2016-04-03:
Well [this](https://blog.wia.io/installing-node-js-on-a-raspberry-pi-3) guy installs the armv6l on a Pi 3 so I guess armv7l should be fine but it appears to be a armv8 chip.

MichMich, 2016-04-03:
I think it's the same. AFAIK the OS is also the same for 3&2

paviro, 2016-04-03:
Are you sure? I mean you can install the same image on all three Pis can't you? Just have to update the kernel. At least I could just put my Pi 1 card into a Pi 2 and just had to update the kernel before.

djsunrise19, 2016-04-03:
@paviro , thanks for your instruction. unfortunately the same npm err in console. here is the logfile:
[npm-debug.txt](https://github.com/MichMich/MagicMirror/files/201405/npm-debug.txt)

paviro, 2016-04-03:
Which distribution are you running and which Pi?

djsunrise19, 2016-04-03:
Raspbian Jessie on Pi2 Model B

paviro, 2016-04-03:
Did you really start it in a window session? I just started it via SSH and got exactly the same error.

djsunrise19, 2016-04-03:
I'm working via putty and SSH.

paviro, 2016-04-03:
As stated in my comment:

> **Important:** `npm start` does NOT work via SSH you have to execute it in a terminal session running in a window-manager.

djsunrise19, 2016-04-03:
Ouch, I'm sorry. I didn't see that comment! It works!

paviro, 2016-04-03:
No problem, glad I could help :)

MadMathWizard, 2016-04-03:
Just for informational purposes, is there a particular reason it does not work in SSH?

And thanks for the help. I’m going through the install notes now .

paviro, 2016-04-03:
There is no window system in your SSH session, same way you can't start other GUI programs via an SSH connection, if you do not connect via `-X` argument. You can also export a display, see [here](http://askubuntu.com/questions/47642/how-to-start-a-gui-software-on-a-remote-linux-pc-via-ssh).

MadMathWizard, 2016-04-03:
Oh right. That makes perfect sense. That is me being silly in my thoughts.
Works great now! Thanks.

MichMich, 2016-04-03:
@djsunrise19 and @madmathwizzard, if You find any bugs in the beta Please open well documented Issues. It would really help me in the Development. 

Can I close this issue?

MadMathWizard, 2016-04-03:
For me you may close it. Thank you.

djsunrise19, 2016-04-03:
Yes, it works.

closed 2016-04-03 by MichMich