MagicMirrorOrg/MagicMirror #133: International Characters

Domi-G, 2016-04-04:
[image]

The version of Roboto that is used right now does only include Latin characters. This helps with keeping the file size of the fonts small, but looks really odd especially if only single non-latin characters are used.

[image]

For testing purposes i have converted the font manually, but using the whole character set leads to huge files:
[image]

One solution for this would be to use the *Google Font API* which uses font subsets to minimize overhead.
But there are issues with this approach:
* The font can’t be loaded while offline
* Privacy concerns
* Unnecessary loading of information over the internet, that could be stored offline

These could be solved by caching or downloading the fonts and CSS files from the Google API, but Google explicitly states that their APIs must not be used in this manner.

Right now I'm working a tool to create subsets easily. I don't want to use fontsquirrel or similar sites because they aren't as customizable and are a bit tedious to use for subsetting.

Do you have any thoughts on the issue? This mostly affects names but also languages using other alphabets.

CFenner, 2016-04-04:
I think as it's just a local site, the size doesn't really matter, does it?

Domi-G, 2016-04-04:
Yeah it's entirely possible that I might be overthinking this. But also keep in mind that 3-4 fonts are loaded:
- robotothin
- roboto_condensedlight
- roboto_condensedregular
- roboto_condensedbold
That adds up to 2MB and considering that the Raspberry Pi has very limited resources it might matter. But I haven't tested it yet.

MichMich, 2016-04-04:
Hi @Domi-G, thanks for opening the issue. Yesterday I worked on some encoding issues (#125 #126) in which I tested the german characters. They seemed to work fine on my end, so I wonder if there is an other issue.

If it's indeed an issue with the font, I think it would not be a problem to lead to full fontset. Since most RPi's wil be used for just the mirror, I don't think memory really will be an issue. Please send a pull request if you can help with this.

Thanks!

MichMich, 2016-04-05:
Should be solved now.

commit 4b8d44aa (2016-04-05): Cleanup fonts, larger charset - fixes #133
commit 700ebc53 (2016-04-05): Merge pull request #135 from Domi-G/v2-beta

Font Improvements - closes #133
closed 2016-04-05 by MichMich