MagicMirrorOrg/MagicMirror #140: Automatic Installer: SHA256SUM issue

djsunrise19, 2016-04-05:
Hi,

I think there is a problem with the automatic installer (at point 4):

`./install.sh: 37: [: 3f7524d3db60175c2323bb2a0a13ad1ca7d47d4ede6f42834b6b8425be70e0a2: unexpected operator
Node.js was downloaded, but verification failed. Make sure sha256sum works.` 

[sha256_issue.txt](https://github.com/MichMich/MagicMirror/files/204991/sha256_issue.txt)

Regards
Maurice

MichMich, 2016-04-05:
@nhubbard, care to look at this?

nhubbard, 2016-04-05:
Interesting, @djsunrise19. I will take a look at this and fix it. Thanks for reporting it!

nhubbard, 2016-04-05:
@djsunrise19, can you try the new version of the script in the latest v2-beta branch? It should work now.

MichMich, 2016-04-06:
Hi @nhubbard, see this issue: https://github.com/MichMich/MagicMirror/issues/127#issuecomment-206113039 - It appears there still are some problems with the installer script.

djsunrise19, 2016-04-06:
@nhubbard , I'll check it tonight, thanks (directly on raspi, no SSH).

MichMich, 2016-04-06:
I appears the latest version was changed (from 5.10.0 to 5.10.1), i've updated the installer accordingly and the now installer seems to work on a clean jessie install. (Boy, does the npm install take long on a RPi2 ... ;))

The only issue I have with using the installer is that changing the file mode for the installer makes the folder dirty. Preventing you from updating using git pull.

CFenner, 2016-04-06:
what aboud something similar to the config file. Make a 'sample' and ignore the original file. In the install the 'sample' can be copied to the one you want to execute.

MichMich, 2016-04-06:
Honestly, I think it could be simpeler. Even just an install script executed from an url.

MichMich, 2016-04-06:
This would work:
````
curl -L https://raw.githubusercontent.com/MichMich/MagicMirror/v2-beta/install.sh | bash
````
We could even put the git clone in there. The installation would be just one command ...

delize, 2016-04-07:
The problem is with the node.js not, with the script itself. 

/usr/local/bin/node: 18: /usr/local/bin/node: Syntax error: ")" unexpected

Plus a bunch of other stuff that is being thrown out when running:

npm config set loglevel info

MichMich, 2016-04-07:
The most recent version of the installer uses `uname -m` to define the architecture. I currently don't have a RPi3 at hand, but with the current script, the right node should be loaded.

nhubbard, 2016-04-08:
Sorry, I've been away for awhile. But yeah, I *thought* I saw on the RPi forums that the RPi 3 was arm64, but I couldn't find a definite answer. That's why I went with arm64, because some were saying it was arm64 and some said it was arm7l.

djsunrise19, 2016-04-08:
@nhubbard , a few minutes ago I tested the installation again on a clean install (raspi 3 this time) and everything works fine. As I can see it's the ARMv7 32 Bit Version.

MichMich, 2016-04-08:
@djsunrise19: does electron work for you?

djsunrise19, 2016-04-08:
@MichMich Yes, everything is fine. I made my config file and start the app as described in the instruction.

MichMich, 2016-04-08:
Cool, Nice to hear.

closed 2016-04-08 by MichMich