MagicMirrorOrg/MagicMirror #144: Content Auto-Login for Cal zabbix etc.

konzeptplus-gmbh, 2016-04-06:
Hi, 

@MichMich Thanks for your great Codes. 
Do you have an idea how I can add a calender with an auto-login? 
As I'm not a Java-scripter I do not know how I should add the html code in the Java. 

```
<HTML>
<HEAD>
<TITLE>Cal</TITLE>
<script>
    function loginForm() {
        document.myform.action = "https://ox.hostkonzept.ch/ox6/#m=calendar&f=26";
        document.myform.submit();
    }
</script>
</HEAD>
<BODY onLoad="loginForm()">
    <FORM NAME="myform" METHOD="POST">
        <INPUT TYPE="hidden" NAME="email" VALUE="myemail"> 
        <INPUT TYPE="hidden" NAME="password" VALUE="mypassw">
    </FORM>
</BODY>
</HTML>
```

Thanks for the help. :-)

CFenner, 2016-04-06:
maybe you could try to add the url like this: `https://USER:PASSWORD@ox.hostkonzept.ch/ox6/#m=calendar&f=26`

konzeptplus-gmbh, 2016-04-06:
I already tied this. Unfortunately this is not working. :-(

MichMich, 2016-04-06:
That probably didn't work because that's the way to do http authentication, and not authentication via POST. The modules does only support GET request, but if you are lucky, the following url could work:
````
https://ox.hostkonzept.ch/ox6/#m=calendar&f=26&email=youremail&password=yourpassword
````

If this doesn't work, you can make a copy of the calendar module, and modify it so that the node_helper uses post requests.

konzeptplus-gmbh, 2016-04-06:
Thanks for the help. Unfortunately it did not work with the url. 

I will try it with the copy of the cal-module. 
Perhaps it would be great to have some infos about how to :-) 
I think it would be good to have the login - post on diffrent modules. 

I'm thinking about a Cam-modul or a Zabbix-alert-modul. :-)  
Regards

MichMich, 2016-04-06:
I still need to write the documentation. But looking around in the helper might give you some clues... 

If you manager to get it up and running with the possibility of configuration. Feel free to send a pull request so other can benefit as wel. 

I close this issue for now. Feel free to reopen if necessary.

closed 2016-04-06 by MichMich