MagicMirrorOrg/MagicMirror #145: Electron not starting on Raspberry Pi

MichMich, 2016-04-06:
I just installed a Raspberry Pi with a clean Jessie installation to give MM2 a try.
Unfortunately, the app will not start, and doesn't give any errors of feedback.

See image below. Any ideas, @paviro & @nhubbard?

[image]

paviro, 2016-04-06:
Strange, maybe try `npm config set log level silly`?

MichMich, 2016-04-07:
[image]

No real helpful information.

MichMich, 2016-04-07:
Ive tried the [Electron Quick Start](https://github.com/electron/electron-quick-start.git), but that seems to have the same issue. I opened an issue in the Electron repository: https://github.com/electron/electron/issues/5069

nhubbard, 2016-04-08:
That's odd. Never have I seen a Node app that crashes without a stack trace.

MichMich, 2016-04-08:
It doesn't even seem to crash. It exits with code 0. But it's an Raspberry Issue, since the Electron Quick Start does the same.

paviro, 2016-04-08:
Strange, that it works without any Problems  for me...

MichMich, 2016-04-08:
@paviro Which RPi are you using? 2 or 3?  On the Electron issue they said it had something to do with the RPi type: https://github.com/electron/electron/issues/5069

paviro, 2016-04-08:
Pi 2 with Jessie.
> Am 08.04.2016 um 11:36 schrieb Michael Teeuw <notifications@github.com>:
> 
> electron/electron#5069 <https://github.com/electron/electron/issues/5069>

MichMich, 2016-04-08:
That's strange indeed. I'm using the same Pi. :-/

paviro, 2016-04-08:
Yeah figured so, that's what I don't get... Does not make any sense.

MichMich, 2016-04-08:
@pavario, so I did another reinstall. Again without succes. Besides the magic mirror installer, I did not run any other installers.

I'm running: **2016-03-18-raspbian-jessie**
Node: **v5.10.1**
GPU memory: **256MB**

Any differences between the above and your situation?

paviro, 2016-04-08:
Seems to be the same thing I did. Well I installed `node` via `apt-get` before but removed it again.

MichMich, 2016-04-08:
Same problem with v4.0.0. 

But how can it be that you are running v4.0.0 if you used the installer?

paviro, 2016-04-08:
There was no installer when I installed it but I did the same things the installer does just with the older node version.

MichMich, 2016-04-08:
If you have an empty card laying around, would you mind trying the installer on a clean jessie installation?

paviro, 2016-04-08:
I'll see if I find one :)

MichMich, 2016-04-08:
Fun fact! I just put the same SD-card into a Pi3. Tried `npm start` and it just worked! 
So there seems to be an issue with the Pi2.

pvyParts, 2016-04-09:
in regards to the installer, this is how i installed node ( before i realized there was an installer )
`curl -sL https://deb.nodesource.com/setup_5.x | sudo -E bash -
`
`sudo apt-get install -y nodejs
`
[from here](http://thisdavej.com/beginners-guide-to-installing-node-js-on-a-raspberry-pi/)

MichMich, 2016-04-09:
@pvyParts: and does the MM run on a RPi2 in your case?

pvyParts, 2016-04-09:
it did when i had one (read till few days ago) i blew mine up tho unfortunately so i'm on a 3 now.

MichMich, 2016-04-09:
Interesting! Thanks for letting me know.

Pukinator, 2016-04-10:
I have the same issue in a Pi B+ running Jessie

MichMich, 2016-04-10:
I tried it yesterday on a different Pi 2, and that one worked. I've got no clue yet on the reason.

nhubbard, 2016-04-10:
Could the Pi be bad? We might want to involve @raspberrypi to see if the issues are related.

MichMich, 2016-04-10:
Nope, it seems like a electron issue. If you run node serveronly there is no issue.

Bangee44, 2016-04-14:
I have the same problems with the PI 1..

MichMich, 2016-04-15:
Ok! Issue solved! It appeared that the non working Pi 2 was actually a Pi 1 B+ (They look very similar!)

In other words: Electron does not work on Pi 1, but does work without any issues on Pi 2 & 3.

closed 2016-04-15 by MichMich

Pukinator, 2016-04-15:
So what should I do when I own a Pi B+?

MichMich, 2016-04-15:
You can start the app using `node server only`, this will start the app, but it will not launch a browser. You'll need to setup a browser solution yourself (like with MagicMirror v1).

Pukinator, 2016-04-15:
Ok I'll try that

desq42, 2016-04-16:
Hi. 

I´m having an issue starting "electron" on my Raspberry Pi 2 (Jessie);

npm ERR! Linux 4.4.7-v7+
npm ERR! argv "/usr/local/bin/node" "/usr/local/bin/npm" "start"
npm ERR! node v5.10.1
npm ERR! npm  v3.8.3
npm ERR! code ELIFECYCLE
npm ERR! magicmirror@2.0.0 start: `electron js/electron.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the magicmirror@2.0.0 start script 'electron js/electron.js'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the magicmirror package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     electron js/electron.js
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs magicmirror
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls magicmirror
npm ERR! There is likely additional logging output above.

Has anybody a clue what am I doing wrong?

Thanks,

desq

MichMich, 2016-04-16:
Could you try

    node serveronly 

To see if that works?

jferch, 2016-04-16:
I believe I had the same error when the automatic magicmirror2 installer did not finish the setup for some reason. I installed node.js for [ARMv7](https://nodejs.org/en/download/stable/) manually which worked for me then. Not sure if that's the case for you as well, @desq42 ?!

desq42, 2016-04-16:
node serveronly works. electron seems so have an issue with my raspberry pi 2 (it´s really a 2).
I´m going to try jferch solution later today.

desq42, 2016-04-17:
no cigar. even did a complete reinstall. I´ll have to live with "serveronly".

MichMich, 2016-04-17:
So you tried: clean jessie install and installer script, and that didn't work?

crazywolf132, 2016-05-15:
Hey. As of today, I am on a RPI 2. It is defenetally a 2. I am running wheezy and everything is fine in `serveronly` ...But when i try to just run `npm start` it fails

MichMich, 2016-05-15:
Please try Jessie

krantni, 2016-05-19:
I had along the same issue and have a I have a Pi 2, Node v6.1.0, npm v3.8.6, full version of Jessie. I noticed at the beginning of the npm install I got:
`npm WARN prefer global colorguard@1.2.0 should be installed with -g`
`npm WARN prefer global coffee-script@1.10.0 should be installed with -g`

I then ran:
`sudo npm install -g coffee-script`
`sudo npm install -g colorguard`
And it got rid of all the warnings. Now the only way `npm start` works is by using `startx` going into the Terminal and running it from there.

matthiasschroeder, 2016-05-21:
Same for me. I am running Jessie on a raspberry pi 3.

masters1222, 2016-06-06:
Hi just wanted to report that I am running a pi 3 with Jesse and am having the same exact issue.

markbarton, 2016-06-08:
Just want to add my report as well - just got a pi 3 and have the same problem Electron will not start - runs fine if I run startx and then use the command console.

MichMich, 2016-06-08:
> runs fine if I run startx and then use the command console.

Could you elaborate on this? 

How do you start the mirror? Which command? Via SSH or Pi Terminal? Jessie or Jessie Lite? Which Node version? How did you install Magic Mirror. What do the logs say?

markbarton, 2016-06-08:
Hi Mich - sorry a bit of a newbie to the pi but ok with NodeJS so bear with me ;-)

I installed using SSH and attempted to run Magic Mirror via SSH - using the display command.

This is what I got:

npm ERR! Linux 4.4.9-v7+
npm ERR! argv "/usr/bin/nodejs" "/usr/bin/npm" "start"
npm ERR! node v6.2.1
npm ERR! npm  v3.9.3
npm ERR! code ELIFECYCLE
npm ERR! magicmirror@2.0.0 start: `electron js/electron.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the magicmirror@2.0.0 start script 'electron js/electron.js'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the magicmirror package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     electron js/electron.js
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs magicmirror
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls magicmirror
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/pi/MagicMirror/npm-debug.log

I then tried npm start from the command console directly on the Pi (when it first boots) and got the same error.

If I launch the GUI using startx and the open a command prompt in there and use the npm start it works fine.

nodeserver only works fine whereever I run it.

So its only electron running from either a SSH session or the command prompt thats the problem - do you think its a security / privileges issue?

markbarton, 2016-06-08:
Heres the npm-debug.log contents

0 info it worked if it ends with ok
1 verbose cli [ '/usr/bin/nodejs', '/usr/bin/npm', 'start' ]
2 info using npm@3.9.3
3 info using node@v6.2.1
4 verbose run-script [ 'prestart', 'start', 'poststart' ]
5 info lifecycle magicmirror@2.0.0~prestart: magicmirror@2.0.0
6 silly lifecycle magicmirror@2.0.0~prestart: no script for prestart, continuing
7 info lifecycle magicmirror@2.0.0~start: magicmirror@2.0.0
8 verbose lifecycle magicmirror@2.0.0~start: unsafe-perm in lifecycle true
9 verbose lifecycle magicmirror@2.0.0~start: PATH: /usr/lib/node_modules/npm/bin/node-gyp-bin:/home/pi/MagicMirror/node_modules/.bin:/usr/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
10 verbose lifecycle magicmirror@2.0.0~start: CWD: /home/pi/MagicMirror
11 silly lifecycle magicmirror@2.0.0~start: Args: [ '-c', 'electron js/electron.js' ]
12 silly lifecycle magicmirror@2.0.0~start: Returned: code: 1  signal: null
13 info lifecycle magicmirror@2.0.0~start: Failed to exec start script
14 verbose stack Error: magicmirror@2.0.0 start: `electron js/electron.js`
14 verbose stack Exit status 1
14 verbose stack     at EventEmitter.<anonymous> (/usr/lib/node_modules/npm/lib/utils/lifecycle.js:245:16)
14 verbose stack     at emitTwo (events.js:106:13)
14 verbose stack     at EventEmitter.emit (events.js:191:7)
14 verbose stack     at ChildProcess.<anonymous> (/usr/lib/node_modules/npm/lib/utils/spawn.js:24:14)
14 verbose stack     at emitTwo (events.js:106:13)
14 verbose stack     at ChildProcess.emit (events.js:191:7)
14 verbose stack     at maybeClose (internal/child_process.js:852:16)
14 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:215:5)
15 verbose pkgid magicmirror@2.0.0
16 verbose cwd /home/pi/MagicMirror
17 error Linux 4.4.11-v7+
18 error argv "/usr/bin/nodejs" "/usr/bin/npm" "start"
19 error node v6.2.1
20 error npm  v3.9.3
21 error code ELIFECYCLE
22 error magicmirror@2.0.0 start: `electron js/electron.js`
22 error Exit status 1
23 error Failed at the magicmirror@2.0.0 start script 'electron js/electron.js'.
23 error Make sure you have the latest version of node.js and npm installed.
23 error If you do, this is most likely a problem with the magicmirror package,
23 error not with npm itself.
23 error Tell the author that this fails on your system:
23 error     electron js/electron.js
23 error You can get information on how to open an issue for this project with:
23 error     npm bugs magicmirror
23 error Or if that isn't available, you can get their info via:
23 error     npm owner ls magicmirror
23 error There is likely additional logging output above.
24 verbose exit [ 1, true ]

MichMich, 2016-06-08:
So just to be sure: when you talk about the "command prompt" on the pi, do you mean the promt it boots into? Or do you mean a terminal window on the Pi after it boots into the GUI?

markbarton, 2016-06-08:
prompt it boots into - doesnt work
command prompt window in gui - works fine

MichMich, 2016-06-08:
That's the expected result. Electron (In which MagicMirror runs) is a GUI application, so it needs a GUI environment. It can't start that environment for you.

markbarton, 2016-06-08:
Ah - thanks.

Seems obvious in hindsight - thats what the Auto start instructions were for.

sundaytill, 2016-06-18:
markbarton/michmich 
     I have the same bug ,  i noticed there is no  config directory ( MagicMirror/config/ ) from the boot terminal. I did install the config directory by ssh.  I am able to npm start from the terminal after  the startX command from the host machine and there is a MagicMirror/config directory on the host machine
How do I fix this ?  Is this an autostart problem ?  A directory issue? I'm really new at all of this please excuse my lack of experience and not understanding  these posts

markbarton, 2016-06-18:
Hi,

It wasnt a bug - it was my lack of understanding :-)

You need to use PM2 - which is a nodejs Package Manager to provide the
autostart capability for the Magic Mirror - https://github.com/Unitech/pm2

Its pretty straight forward to use.

Mark

MichMich, 2016-06-18:
@sundaytill This doesn't seem related to this issue. Please open a support question in http://forum.magicmirror.builders 

Thanks!

sundaytill, 2016-06-18:
Hi
    Mark ,
Thank you

Serubin, 2016-10-20:
Any update on the issue in the op? Seems the thread got a bit sidetracked ;)

MichMich, 2016-10-20:
Yeah the OP is solved a while ago. Had something to do with an older Electron version. Not relevant anymore.

blindarrow, 2016-10-22:
Still getting the issue on a Raspberry Pi 3. The auto-install script happens to mention this as well:

```
Installing dependencies ...
npm WARN deprecated electron-prebuilt@1.3.8: electron-prebuilt has been renamed to electron. For more details, see http://electron.atom.io/blog/2016/08/16/npm-install-electron
npm WARN prefer global coffee-script@1.10.0 should be installed with -g
npm WARN prefer global colorguard@1.2.0 should be installed with -g
npm WARN prefer global jsonlint@1.6.2 should be installed with -g

> electron-prebuilt@1.3.8 postinstall /home/pi/MagicMirror/node_modules/electron-prebuilt
> node install.js
```

I then ran...

```
npm install -g coffee-script
npm install -g colorguard
npm install -g jsonlint
npm install -g electron

Still getting the issue. This is on a Raspberry Pi 3 with a clean install of Raspbian, all updated.

Full error on doing npm start in /MagicMirror..

```
pi@raspberrypi:~/MagicMirror $ npm start

> magicmirror@2.0.0 start /home/pi/MagicMirror
> electron js/electron.js

npm ERR! Linux 4.4.26-v7+
npm ERR! argv "/usr/bin/nodejs" "/usr/bin/npm" "start"
npm ERR! node v6.9.1
npm ERR! npm  v3.10.8
npm ERR! code ELIFECYCLE
npm ERR! magicmirror@2.0.0 start: `electron js/electron.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the magicmirror@2.0.0 start script 'electron js/electron.js'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the magicmirror package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     electron js/electron.js
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs magicmirror
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls magicmirror
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/pi/MagicMirror/npm-debug.log
pi@raspberrypi:~/MagicMirror $ 
```

MichMich, 2016-10-22:
Could you give this repo a try to see if electron works at all? https://github.com/electron/electron-quick-start

blindarrow, 2016-10-22:
Same results it looks like:

```
npm ERR! Linux 4.4.26-v7+
npm ERR! argv "/usr/bin/nodejs" "/usr/bin/npm" "start"
npm ERR! node v6.9.1
npm ERR! npm  v3.10.8
npm ERR! code ELIFECYCLE
npm ERR! electron-quick-start@1.0.0 start: `electron .`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the electron-quick-start@1.0.0 start script 'electron .'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the electron-quick-start package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     electron .
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs electron-quick-start
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls electron-quick-start
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/pi/electron-quick-start/npm-debug.log
pi@raspberrypi:~/electron-quick-start $ 
```

Results from the npm-debug.log:

```
pi@raspberrypi:~/electron-quick-start $ cat /home/pi/electron-quick-start/npm-debug.log
0 info it worked if it ends with ok
1 verbose cli [ '/usr/bin/nodejs', '/usr/bin/npm', 'start' ]
2 info using npm@3.10.8
3 info using node@v6.9.1
4 verbose run-script [ 'prestart', 'start', 'poststart' ]
5 info lifecycle electron-quick-start@1.0.0~prestart: electron-quick-start@1.0.0
6 silly lifecycle electron-quick-start@1.0.0~prestart: no script for prestart, continuing
7 info lifecycle electron-quick-start@1.0.0~start: electron-quick-start@1.0.0
8 verbose lifecycle electron-quick-start@1.0.0~start: unsafe-perm in lifecycle true
9 verbose lifecycle electron-quick-start@1.0.0~start: PATH: /usr/lib/node_modules/npm/bin/node-gyp-bin:/home/pi/electron-quick-start/node_modules/.bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/local/games:/usr/games
10 verbose lifecycle electron-quick-start@1.0.0~start: CWD: /home/pi/electron-quick-start
11 silly lifecycle electron-quick-start@1.0.0~start: Args: [ '-c', 'electron .' ]
12 silly lifecycle electron-quick-start@1.0.0~start: Returned: code: 1  signal: null
13 info lifecycle electron-quick-start@1.0.0~start: Failed to exec start script
14 verbose stack Error: electron-quick-start@1.0.0 start: `electron .`
14 verbose stack Exit status 1
14 verbose stack     at EventEmitter.<anonymous> (/usr/lib/node_modules/npm/lib/utils/lifecycle.js:255:16)
14 verbose stack     at emitTwo (events.js:106:13)
14 verbose stack     at EventEmitter.emit (events.js:191:7)
14 verbose stack     at ChildProcess.<anonymous> (/usr/lib/node_modules/npm/lib/utils/spawn.js:40:14)
14 verbose stack     at emitTwo (events.js:106:13)
14 verbose stack     at ChildProcess.emit (events.js:191:7)
14 verbose stack     at maybeClose (internal/child_process.js:877:16)
14 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:226:5)
15 verbose pkgid electron-quick-start@1.0.0
16 verbose cwd /home/pi/electron-quick-start
17 error Linux 4.4.26-v7+
18 error argv "/usr/bin/nodejs" "/usr/bin/npm" "start"
19 error node v6.9.1
20 error npm  v3.10.8
21 error code ELIFECYCLE
22 error electron-quick-start@1.0.0 start: `electron .`
22 error Exit status 1
23 error Failed at the electron-quick-start@1.0.0 start script 'electron .'.
23 error Make sure you have the latest version of node.js and npm installed.
23 error If you do, this is most likely a problem with the electron-quick-start package,
23 error not with npm itself.
23 error Tell the author that this fails on your system:
23 error     electron .
23 error You can get information on how to open an issue for this project with:
23 error     npm bugs electron-quick-start
23 error Or if that isn't available, you can get their info via:
23 error     npm owner ls electron-quick-start
23 error There is likely additional logging output above.
24 verbose exit [ 1, true ]
pi@raspberrypi:~/electron-quick-start $ 
```

Sillec123, 2016-10-25:
I have the same error with:
Pi2
electron-v1.4.4

npm ERR! node v6.9.1
npm ERR! npm  v3.10.8

node serveronly is running but PM2 not. 

[STREAMING] Now streaming realtime logs for [all] processes
0|mm       | 2016-10-25T07:16:04.470Z INFO no settings file found, will only use default settings
0|mm       | 2016-10-25T07:16:05.644Z INFO Presets loaded: { example:
0|mm       |    { players:
0|mm       |       [ { roomName: 'Bathroom', volume: 10 },
0|mm       |         { roomName: 'Kitchen', volume: 10 },
0|mm       |         { roomName: 'Office', volume: 10 },
0|mm       |         { roomName: 'Bedroom', volume: 10 },
0|mm       |         { roomName: 'TV Room', volume: 15 } ],
0|mm       |      playMode: { shuffle: true, repeat: 'all', crossfade: false },
0|mm       |      pauseOthers: false,
0|mm       |      favorite: 'My example favorite' } }
0|mm       | 2016-10-25T07:16:05.789Z INFO http server listening on port 5005
/root/.pm2/logs/mm-error-0.log last 10 lines:
0|mm       | npm ERR! Tell the author that this fails on your system:
0|mm       | npm ERR!     electron js/electron.js
0|mm       | npm ERR! You can get information on how to open an issue for this project with:
0|mm       | npm ERR!     npm bugs magicmirror
0|mm       | npm ERR! Or if that isn't available, you can get their info via:
0|mm       | npm ERR!     npm owner ls magicmirror
0|mm       | npm ERR! There is likely additional logging output above.
0|mm       |
0|mm       | npm ERR! Please include the following file with any support request:
0|mm       | npm ERR!     /root/MagicMirror/npm-debug.log

MichMich, 2016-10-25:
@blindarrow Please open an issue in the electron repository.
@Sillec123 Did you try? https://github.com/electron/electron-quick-start if not, please open an issue in the electron repository.

[13 more replies]