MagicMirrorOrg/MagicMirror #149: 404 Not Found

Robiv8, 2016-04-08:
I have this error, what can i do?

robi@robis-thinkpad:~/MagicMirror$ sudo ./install.sh
Paketlisten werden gelesen... Fertig
Abhängigkeitsbaum wird aufgebaut.       
Statusinformationen werden eingelesen.... Fertig
build-essential ist schon die neueste Version.
wget ist schon die neueste Version.
curl ist schon die neueste Version.
unzip ist schon die neueste Version.
0 aktualisiert, 0 neu installiert, 0 zu entfernen und 3 nicht aktualisiert.
--2016-04-08 18:32:12--  https://nodejs.org/dist/latest/node-v5.10.1-linux-x86_64.tar.gz
Auflösen des Hostnamen »nodejs.org (nodejs.org)«... 104.20.23.46, 104.20.22.46, 2400:cb00:2048:1::6814:172e, ...
Verbindungsaufbau zu nodejs.org (nodejs.org)|104.20.23.46|:443... verbunden.
HTTP-Anforderung gesendet, warte auf Antwort... 404 Not Found
2016-04-08 18:32:12 FEHLER 404: Not Found.

MichMich, 2016-04-08:
It seems it can't find the file:
````
https://nodejs.org/dist/latest/node-v5.10.1-linux-x86_64.tar.gz
````

This filename is generated based on the architecture you're currently running. What Raspberry Pi and Operating System are you running?

Robiv8, 2016-04-08:
I try it on a ThinkPad with Unbuntu 15.10 Intel® Core™ i5-2520M CPU @ 2.50GHz × 4
uname -m
x86_64

MichMich, 2016-04-08:
The installer is Raspberry only. On Ubuntu, make sure the latest Node.js is installed. And then run: 'npm install' from the MagicMirror folder. 

After the installation you can start the app via 'npm start'.

closed 2016-04-08 by MichMich

Robiv8, 2016-04-08:
Great! Thank you

Robiv8, 2016-04-08:
1. Question
How can i restart the mirror on the Desktop after going out with alt?

MichMich, 2016-04-08:
npm start

Robiv8, 2016-04-08:
hmm.. on the wrong place?

0 info it worked if it ends with ok
1 verbose cli [ '/usr/local/bin/node', '/usr/local/bin/npm', 'start' ]
2 info using npm@3.8.3
3 info using node@v5.10.1
4 verbose config Skipping project config: /home/robi/.npmrc. (matches userconfig)
5 verbose stack Error: ENOENT: no such file or directory, open '/home/robi/package.json'
5 verbose stack     at Error (native)
6 verbose cwd /home/robi
7 error Linux 4.2.0-35-generic
8 error argv "/usr/local/bin/node" "/usr/local/bin/npm" "start"
9 error node v5.10.1
10 error npm  v3.8.3
11 error path /home/robi/package.json
12 error code ENOENT
13 error errno -2
14 error syscall open
15 error enoent ENOENT: no such file or directory, open '/home/robi/package.json'
16 error enoent ENOENT: no such file or directory, open '/home/robi/package.json'
16 error enoent This is most likely not a problem with npm itself
16 error enoent and is related to npm not being able to find a file.
17 verbose exit [ -2, true ]

Robiv8, 2016-04-08:
OK STOP !
I found it ".. run from the MagicMirror Folder" 
Is there a way to make a Desktop Icon with this Command to Open the Mirror from Desktop without the Terminal?

MichMich, 2016-04-08:
No idea yet ... If you find out how, please let us know so we can add it to the readme. (or send a pull request.)

paviro, 2016-04-08:
Should be possible with a .desktop file. Google something like ".desktop run shell  script" for details, but I have not tried it.

Robiv8, 2016-04-08:
Thank you. I try this and its work! :-)
Create a yourname.desktop file in your /home/user/Desktop/ directory that looks like this:

THIS IS MY EXAMPLE
PUT YOUR OWEN PATH AND NAME INTO

`[Desktop Entry]
Name=MagicMirror
Comment=Startet den Spiegel
Exec=/bin/bash -c "cd /home/robi/MagicMirror&&npm start"
Icon=/home/robi/MagicMirror/mirror.png
Type=Application
Terminal=false`

Thanks for help.

MichMich, 2016-04-08:
Interesting! Thanks. I'll create a MagicMirror Icon soon, and add the above file to the repo so anyone can use it.

MichMich, 2016-04-08:
FYI, the following file would work for most people:

````
[Desktop Entry]
Name=MagicMirror
Comment=Start MagicMirror
Exec=/bin/bash -c "cd ~/MagicMirror&&npm start"
Icon=~/MagicMirror/mirror.png
Type=Application
Terminal=false
````