MagicMirrorOrg/MagicMirror #150: v2 Branch - Application force closes after 12 - 24 hours of running

delize, 2016-04-09:
Hello,

I have noticed that after running for v2 Branch for extended periods of time (basically anything longer than 10 hours), the application decides to force quit, itself.

It quits with the following lines in the console when running "npm start":
```
~/Downloads/MagicMirror $ npm start
npm info it worked if it ends with ok
npm info using npm@3.8.3
npm info using node@v5.10.1
npm info lifecycle magicmirror@2.0.0~prestart: magicmirror@2.0.0
npm info lifecycle magicmirror@2.0.0~start: magicmirror@2.0.0

> magicmirror@2.0.0 start /home/pi/Downloads/MagicMirror
> electron js/electron.js

Loading config ...
Loading module helpers ...
No helper found for module: alert.
No helper found for module: clock.
Initializing new module helper ...
No helper found for module: compliments.
No helper found for module: currentweather.
No helper found for module: weatherforecast.
Initializing new module helper ...
All module helpers loaded.
Starting server op port 8080 ... 
Server started ...
Connecting socket for: calendar
Starting node helper for: calendar
Connecting socket for: newsfeed
Starting module: newsfeed
Sockets connected & modules started ...
Launching application.
(OMIT CALENDER DETAILS)
[]
[]
npm info lifecycle magicmirror@2.0.0~poststart: magicmirror@2.0.0
npm info ok 

```

I have tested this on the lastest version I had which was yesterday, but due to the amount of time required to make this occur, I can't be on the most up to date version constantly. I have done a git pull now, and will test again  with todays code release.

MichMich, 2016-04-09:
Thanks for reporting! It doesn't seem to close with an error, so hard to debug. I'll put the current version in my real mirror today, so I can start long-time testing. I'm running it using PM2 so it automatically recovers after a failure. 

Keep me posted on your findings.

pvyParts, 2016-04-10:
i have had this happen as well, nothing in the log. I've relaunched with silly log level to see if anything turns up. (so far been running for about 14 hours)

could you elaborate on how to setup PM2 to run MM.

MichMich, 2016-04-10:
Check out: https://github.com/MichMich/MagicMirror/wiki/Auto-Starting-MagicMirror

MichMich, 2016-04-10:
I got mine up and running now for 30 hours.

paviro, 2016-04-10:
Nice will test your autostart method :)

MichMich, 2016-04-10:
It's very handy, especially since it allows you to easily start, stop and restart your mirror via SSH.

MichMich, 2016-04-11:
Running for 47 hour steady now. @pvyParts, how is your mirror running?

pvyParts, 2016-04-11:
it hasn't stopped since my last comment. which i guess is a good thing lol

MichMich, 2016-04-11:
Good to hear! :)

MichMich, 2016-04-11:
Mine appeared to be restarted 4 hours ago. After 53 hours of uptime. No clue what happened. No signs of errors in the logs.

MichMich, 2016-04-15:
Hi @delize and @pvyParts, any news in this issue?

pvyParts, 2016-04-15:
Mine is still running. Haha, I've actually lost count now. so i'm guessing it is pretty random. i'll keep my mirror on silly log level tho and if it does stop let you know.

nhubbard, 2016-04-17:
If I run `node serveronly` for a while, I will eventually get a `ECONNRESET` error from Express. This could possibly be the cause.

MichMich, 2016-04-17:
Thanks for the info! Could you share the full log?

nhubbard, 2016-04-18:
Here is the basic gist of what happens:
```javascript
events.js:72
    throw er; // Unhandled 'error' event
          ^
Error: read ECONNRESET
at errnoException (net.js:904:11)
at Pipe.onread (net.js:558:19)
```

MichMich, 2016-04-18:
I pushed an update this morning which probably solves this issue: https://github.com/MichMich/MagicMirror/commit/bef5b48fce182c4ed1025eee91ab1ff0c14d44ac

Could you please upgrade to the latest version and try again?

MichMich, 2016-04-19:
The core app now includes a catch all for all exceptions. So this issue should be solved. 

I close this issue for now, but feel free to let me know if I need to reopen the issue.

closed 2016-04-19 by MichMich

vanimesh76, 2016-04-23:
Hey Thanks for the codes.
**Electron error:**
while installing electron on Raspberry pi 3 i get this error.

` Failed at the electron-prebuilt@0.37.7 postinstall script 'node install.js'.
npm ERR! This is most likely a problem with the electron-prebuilt package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node install.js
npm ERR! You can get their info via:
npm ERR!     npm owner ls electron-prebuilt
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/pi/MagicMirror/npm-debug.log
` 

**The Whole Error is like this**

`Downloading electron-v0.37.7-linux-arm.zip
Error: read ECONNRESET
/home/pi/MagicMirror/node_modules/electron-prebuilt/install.js:22
  throw err
  ^

Error: read ECONNRESET
    at exports._errnoException (util.js:837:11)
    at TLSWrap.onread (net.js:544:26)
npm ERR! Linux 4.1.19-v7+
npm ERR! argv "/usr/local/bin/node" "/usr/local/bin/npm" "install"
npm ERR! node v4.0.0
npm ERR! npm  v2.14.2
npm ERR! code ELIFECYCLE

npm ERR! electron-prebuilt@0.37.7 postinstall: `node install.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the electron-prebuilt@0.37.7 postinstall script 'node install.js'.
npm ERR! This is most likely a problem with the electron-prebuilt package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node install.js
npm ERR! You can get their info via:
npm ERR!     npm owner ls electron-prebuilt
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/pi/MagicMirror/npm-debug.log
`

**_Any Solution? Please Need Help Immediately, Gotta Gift this to My Girl_**

paviro, 2016-04-23:
Check your internet connection!

MichMich, 2016-04-23:
Update node to v5.11.0

vanimesh76, 2016-04-24:
It wasn't working on my wifi, had to make a HotSpot using 3G,(not working on 4G) :)
your installer automatically updates to the latest Node Js.

also how do i get rid of the notification (Welcom, start was successful)

paviro, 2016-04-24:
See the documentation of the alert module.

vanimesh76, 2016-04-24:
What is an easy way to edit files and modules and get them updated on the raspberry.
or i have to just edit it on the raspberry to see the changes?

delize, 2016-04-24:
@MichMich 

Hey, unfortunately, I am still having the same problems. I don't have logs on this one, as it was setup to automatically start after boot, but it did crash sometime between 2:00 AM and 8:30 AM after I woke up. 

This was after pulling the latest files yesterday (from around 4:00 PM PT), and reconfiguring the setup (it had been about a month since I had changed anything). Maybe it was just the way that I started the MagicMirror so I will give it a shot with the pm2 way this time.

But it definitely crashed by itself. 

Is there any way that you can possibly have it output logfiles in the working directory of whever npm start is run, so that there is a constant place for log files? Or maybe there is a place where npm stores logfiles, and I don't know about it.

For record I was on bc2f2f5 when this occured, I have no just upgraded to 4338cb3.

MichMich, 2016-04-24:
@vanimesh76 You only need to edit the config file. You need to do that on the Raspberry.

@delize: you can use `pm2 logs mm` to see the logs.

delize, 2016-04-24:
@MichMich I just started using pm2 after the crash. I would expect the logs would not show there unless I was using PM2 from the start, wouldn't that be correct?

MichMich, 2016-04-24:
Indeed.

MichMich, 2016-04-24:
`pm2 show mm` is also very helpful. It shows the amount of 'unstable restarts'.

Danesz, 2016-05-07:
Hi @MichMich ,

Question about your commit:

> I pushed an update this morning which probably solves this issue: #bef5b48

I see you catch all the exceptions, that's ok, so there will be no more exception popup on the screen, and maybe no more crash.
But will the modules reload the content after a caught exception?
Because basically I have never had a crash, I just saw a "frozen" UI with an exception popup.

I mean maybe it is a better idea to catch the uncaughtException, then exit, and pm2 will restart the whole app nice and clean.

MichMich, 2016-05-07:
The reason I built it in, was that a lot of crashes were caused by the fact the internet connection was down (dns exception). Maybe we should check if that's the reason, and if not: restart. I must admit it was a bit of a hot-fix. So a PR with a more sophisticated solution is welcome.

paviro, 2016-05-07:
In Python you can name exceptions and only continue running in some cases, is that possible in JavaScript?

MichMich, 2016-05-07:
You get a error back which you can evaluate.

Danesz, 2016-05-07:
I have just checked your module documentation.
What if you catch all the exceptions in your main class, and send a notification to the modules or call a "restart" method (which is currently not yet implemented) in the modules to restore their state?
Or if you can identify in your main class the root cause of the exception (which module), then you can just restart that module.

abtin, 2016-05-13:
I have experienced the same issue, almost consistently. I'm running MM2 in Raspi 3 Model B. I just pulled from the master and restarted the mirror. Will submit the logs if it crashes again. BTW, great jobs everyone.

abtin, 2016-05-18:
Since I've executed the following update commands, MM2 hasn't crashed. It's been running for more than 54 hours now.
sudo apt-get update
sudo apt-get upgrade
sudo apt-get dist-upgrade
sudo apt-get clean
sudo rpi-update

amolpagar, 2017-04-26:
Hi MichMich

I reached to till below point  after complete installation, I see a black screen on Pi, none of the applications are loaded, can you please let me know what could be the issue?

pi@raspberrypi ~/MagicMirror $ npm start

> magicmirror@2.1.1 start /home/pi/MagicMirror
> sh run-start.sh

Starting MagicMirror: v2.1.1
Loading config ...
Loading module helpers ...
No helper found for module: alert.
Initializing new module helper ...
Module helper loaded: updatenotification
No helper found for module: clock.
Initializing new module helper ...
Module helper loaded: calendar
No helper found for module: compliments.
No helper found for module: currentweather.
No helper found for module: weatherforecast.
Initializing new module helper ...
Module helper loaded: newsfeed
All module helpers loaded.
Starting server on port 8080 ... 
Starting server op port 8080 ... 
Server started ...
Connecting socket for: updatenotification
Connecting socket for: calendar
Starting node helper for: calendar
Connecting socket for: newsfeed
Starting module: newsfeed
Sockets connected & modules started ...
Launching application.

amolpagar, 2017-04-26:
https://github.com/MichMich/MagicMirror/issues/222 
I checked at above url and tried removing module one by one, it looks like only newsfeed might work :(

amolpagar, 2017-04-26:
I **cant** see such lines after launching application, does it take too much time to load? looking at interval??

Create new calendar fetcher for url: http://calendar.google.com/calendar/ical/de.german%23holiday%40group.v.calendar.google.com/public/basic.ics - Interval: 300000
Create new news fetcher for url: http://www.tagesschau.de/xml/rss2 - Interval: **300000**

amolpagar, 2017-05-01:
I found a solution after digging a lot, 

The problem was a warning during npm install as below
pi@raspberrypi ~/MagicMirror $ sudo npm install
npm WARN lifecycle magicmirror@2.1.1~install: cannot run in wd %s %s (wd=%s) magicmirror@2.1.1 cd vendor && npm install /home/pi/MagicMirror
npm WARN lifecycle magicmirror@2.1.1~postinstall: cannot run in wd %s %s (wd=%s) magicmirror@2.1.1 sh installers/postinstall/postinstall.sh /home/pi/MagicMirror
npm WARN grunt-stylelint@0.8.0 requires a peer of stylelint@^7.8.0 but none was installed.
npm WARN stylelint-config-standard@16.0.0 requires a peer of stylelint@^7.8.0 but none was installed.
pi@raspberrypi ~/MagicMirror $ ls
CHANGELOG.md          fonts         jsconfig.json  README.md     translations
config                Gruntfile.js  LICENSE.md     run-start.sh  vendor
css                   index.html    modules        serveronly
docker-entrypoint.sh  installers    node_modules   splashscreen
Dockerfile            js            package.json   tests
pi@raspberrypi ~/MagicMirror $ cd vendor/
pi@raspberrypi ~/MagicMirror/vendor $ sudo npm install
magicmirror-vendors@ /home/pi/MagicMirror/vendor
├── font-awesome@4.7.0 
├── moment@2.18.1 
├── moment-timezone@0.5.13 
└── weathericons@2.1.0 

Black screen is defnitely because of config.js file issue. but this time, js file was not loading at the time of execution because of moments being not installed in vendor directory.

In order to find this you will have to use npm start dev command (dont forget to use sudo if you are not running as admin). Once this command is executed, it will open electron browser in debug mode.

check the console log for the errors, there you will find that moments are not being loaded because of reference is not found for js files. for.e.g Clock.js 77 

to to resolve this you will have to execute npm install command inside vendor directory, the only mistake I did was ignoring the warnings during installations of npm. warning clearly says to execute npm install inside vendor directory, when you have it resolved. 

try executing npm start again and it will do a magic as expected !

fewieden, 2017-05-01:
cannot run in wd %s %s (wd=%s)

can be avoided by `npm install --unsafe-perm`

BismaGhaffar, 2017-07-07:
I am continuously getting these errors help me!

C:\Users\Bisma\Desktop\MagicMirror>npm install

> magicmirror@2.1.2 install C:\Users\Bisma\Desktop\MagicMirror
> cd vendor && npm install

npm WARN invalid config loglevel="notice"
up to date in 0.44s

> magicmirror@2.1.2 postinstall C:\Users\Bisma\Desktop\MagicMirror
> sh installers/postinstall/postinstall.sh

'sh' is not recognized as an internal or external command,
operable program or batch file.
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! magicmirror@2.1.2 postinstall: `sh installers/postinstall/postinstall.s
h`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the magicmirror@2.1.2 postinstall script.
npm ERR! This is probably not a problem with npm. There is likely additional log
ging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\Bisma\AppData\Roaming\npm-cache\_logs\2017-07-07T11_36_20_
182Z-debug.log

C:\Users\Bisma\Desktop\MagicMirror>

paviro, 2017-07-07:
The installer is meant for the Raspberry Pi and can only run on Linux.

BismaGhaffar, 2017-07-07:
So,How can I run on my windows

BismaGhaffar, 2017-07-07:
Few month ago i run this on my window via npm start command  easily but now i am getting error what is the solution.

paviro, 2017-07-07:
Use the manual installation guide as reference :)

fewieden, 2017-07-07:
windows cmd cannot run sh by default, take a look here https://forum.magicmirror.builders/topic/2191/won-t-start-on-windows-after-upgrade