MagicMirrorOrg/MagicMirror #151: Please create a config file

Pukinator, 2016-04-09:
When I edit the config.js file and just add my API key everything works fine but as soon as I change the city to mine (Luzern) it tells me that I should create a config file 
The Javascript console of Chrome tells me this: [image]

Here is my config.js:
```
/* exported config */

/* Magic Mirror Config
 *
 * By Michael Teeuw http://michaelteeuw.nl
 * MIT Licensed.
 */

var config = {
	port: 8080,

	language: ‚en‘,
	timeFormat: 24,

	modules: [
		{
			module: 'alert',
		},
		{
			module: 'clock',
			position: 'top_left'
		},
		{
			module: 'calendar',
			header: 'US Holidays',
			position: 'top_left',
			config: {
				calendars: [
					{
						symbol: 'calendar-check-o ',
						url: 'webcal://www.calendarlabs.com/templates/ical/US-Holidays.ics'
					}
				]
			}
		},
		{
			module: 'compliments',
			position: 'lower_third',
		},
		{
			module: 'currentweather',
			position: 'top_right',
			config: {
				location: ‚Luzern‘,
				appid: '44b91dce42d139e6a6a365dc4989953e'
			}
		},
		{
			module: 'weatherforecast',
			position: 'top_right',
			header: 'Weather Forecast',
			config: {
	            location: ‚Luzern‘,
	            appid: '44b91dce42d139e6a6a365dc4989953e'
			}
		},
		{
			module: 'newsfeed',
			position: 'bottom_bar',
			config: {
				feedUrl: 'http://www.nytimes.com/services/xml/rss/nyt/HomePage.xml',
				showPublishDate: true,
			}
		},
	]

};

/*************** DO NOT EDIT THE LINE BELOW ***************/
if (typeof module !== 'undefined') {module.exports = config;}
```
MagicMirror is running on a Pi B+ with RaspBian Jessie

MichMich, 2016-04-09:
Your quotes around ` ‚Luzern‘,` are incorrect.

closed 2016-04-09 by MichMich

Pukinator, 2016-04-09:
Thanks for your answer! TextEdit automatically does that, I deactivated the function now

saireddy1438, 2019-04-07:
does single quote or double quote matter?

ercol10, 2019-12-04:
This is the error that I am getting. Any help would be great !

pi@raspberrypi:~/MagicMirror $ npm start

> magicmirror@2.9.0 start /home/pi/MagicMirror
> sh run-start.sh

Starting MagicMirror: v2.9.0
Loading config ...
WARNING! Could not find config file. Please create one. Starting with default configuration.
Loading module helpers ...
Initializing new module helper ...
Module helper loaded: updatenotification
No helper found for module: helloworld.
All module helpers loaded.
Starting server on port 8080 ... 
Server started ...
Connecting socket for: updatenotification
Sockets connected & modules started ...
Launching application.
Shutting down server...
Stopping module helper: updatenotification

ercol10, 2019-12-04:
Here is my config file 

language: "en",
	timeFormat: 24,
	units: "metric",

	modules: [
		{
			module: "alert",
		},
		{
			module: "updatenotification",
			position: "top_bar"
		},
		{
			module: "clock",
			position: "top_left"
		},
		{
			module: "calendar",
			header: "US Holidays",
			position: "top_left",
			config: {
				calendars: [
					{
						symbol: "calendar-check",
						url: "webcal://www.calendarlabs.com/ical-calendar/ics/76/US_Holidays.ics"					}
				]
			}
		},
		{
			module: "compliments",
			position: "lower_third"
		},
		{
			module: "currentweather",
			position: "top_right",
			config: {
				location: "New York",
				locationID: "",  //ID from http://bulk.openweathermap.org/sample/city.list.json.gz; unzip the gz file and find your city
				appid: "YOUR_OPENWEATHER_API_KEY"
			}
		},
		{
			module: "weatherforecast",
			position: "top_right",
			header: "Weather Forecast",
			config: {
				location: "New York",
				locationID: "5128581",  //ID from http://bulk.openweathermap.org/sample/city.list.json.gz; unzip the gz file and find your city
				appid: "YOUR_OPENWEATHER_API_KEY"
			}
		},
		{
			module: "newsfeed",
			position: "bottom_bar",
			config: {
				feeds: [
					{
						title: "New York Times",
						url: "http://www.nytimes.com/services/xml/rss/nyt/HomePage.xml"
					}
				],
				showSourceTitle: true,
				showPublishDate: true,
				broadcastNewsFeeds: true,
				broadcastNewsUpdates: true
			}
		},
	]

};

/*************** DO NOT EDIT THE LINE BELOW ***************/
if (typeof module !== "undefined") {module.exports = config;}