MagicMirrorOrg/MagicMirror #153: There are some errors in the description

Robiv8, 2016-04-09:
Hello World in your Module readme
    {
        module: 'helloworld',
        position: 'bottom_bar', // This can be any of the regions.
        config: {
            // See 'Configuration options' for more information.
            text: 'Hello world!', **<-- wrong**
        }
    }

Your example config.js on GitHub
   {
            module: "helloworld",
            position: "lower_third",
            classes: "small"
            config: {
                text: "Hello world!" **<-- wrong**
            }
        }

My functioning Module:
{
            module: "helloworld",
            position: "lower_third",
            classes: "small"
            config: {
                text: ’Hello world!’ **<-- correct**
            }
        }

nhubbard, 2016-04-09:
I would recommend that you fork the project, make the change to fix this issue, and then make a pull request to correct it.

MichMich, 2016-04-10:
Is it just me, or isn't there really any difference? All three versions should work, I think.

nhubbard, 2016-04-10:
I would agree. JSCS *might* complain, but there isn't really a difference.

MichMich, 2016-04-10:
I close this issue for now. Feel free to reopen.

closed 2016-04-10 by MichMich

Robiv8, 2016-04-10:
I feel just a bit fooled. I have needed a little while to come to this tiny error. And no! I am not a programmer, coder or the like.
In the first 2 versions "Please create Config" comes.

And sorry, I dont know how everything works here with forks, pull etc ..
So do what you want with it

MichMich, 2016-04-10:
Please, Don't feel fooled. Your contribution is more than welcome. Unfortunately I fail to find the issue in your description. All three versions seem correct. Since you are currently not experiencing any issues (as far as I can understand) I closed this issue.

Robiv8, 2016-04-12:
It is probably too small to read.
I have made the points again in bold.
In 1st apostrophe closed comma.
At 2 quotes
At 3 the correct version. Apostrophe without comma.
As I said are No1. and No.2 from the tutorial and do not work because the message is you should create a config.
Greeting

## 1. text: **'**Hello world!**',** <-- wrong
2. text: **"**Hello world!**"** <-- wrong
3. text: **’**Hello world!**’** <-- correct

MichMich, 2016-04-12:
Honestly, I think it's an issue with your text editor which converts te comma's into the wrong entities. What editor are you using?

The extra comma does not matter since it's a javascript object.

I'm using the code from the tutorial and it works.

Robiv8, 2016-04-12:
Ha Ha, We are talking past each other, but no matter!
No, it is not on my text editor, but no matter.
I give up.
Keep it up. The project is great.

MichMich, 2016-04-12:
Can You send a pullrequest?

Robiv8, 2016-04-12:
OK let me try.

commit bc8dccd4 (2016-04-13): Fix #153