MagicMirrorOrg/MagicMirror #159: clock goes wonky after a few minutes

burnsintx, 2016-04-11:
The system starts up great.  After a few minutes, the clock starts showing a very odd number in the UI.  Right now the clock shows "2.2250717376122993e-308:56".  The only changes I have made to the modules is from 24 to 12 hr clock, and I removed the seconds display by putting in "false".

MichMich, 2016-04-12:
That is really odd, especially since this does nothing more than `moment().format("hh:mm")`. Does this happen all the time? Or is it unpredictable behavior?

burnsintx, 2016-04-12:
Yeah something is weird.  I restarted it a few times, changing the parameters that I changed, seconds and 12/24.  It did it every time after about 1 minute.  The weird thing as well is that it changed the clock display on the "desktop" of raspbian as well.  The settings still show that it is a 12 hour clock, but it displays a 24 hr clock.

MichMich, 2016-04-12:
I think this is an moment.js related issue. Do you have the knowledge to do a technical test with moment.js only?

dthomas888, 2016-04-12:
Is is definitely an issue with moment.js and the pi. I modified the module to verify and this works fine.

    
    var currentTime = new Date ( );
  	var currentHours = currentTime.getHours ( );
  	var currentMinutes = currentTime.getMinutes ( );
  	//var currentSeconds = currentTime.getSeconds ( );

  	// Pad the minutes and seconds with leading zeros, if required
  	currentMinutes = ( currentMinutes < 10 ? "0" : "" ) + currentMinutes;
  	//currentSeconds = ( currentSeconds < 10 ? "0" : "" ) + currentSeconds;

  	// Choose either "AM" or "PM" as appropriate
  	var timeOfDay = ( currentHours < 12 ) ? "AM" : "PM";

  	// Convert the hours component to 12-hour format if needed
  	currentHours = ( currentHours > 12 ) ? currentHours - 12 : currentHours;
    
    //pad hours with leading zero
    currentHours = ( currentHours < 10 ? "0" : "" )  + currentHours;
    
    timeWrapper.innerHTML = currentHours + ":" + currentMinutes

		// Set content of wrappers.
		dateWrapper.innerHTML = moment().format("dddd, LL"); 
		//timeWrapper.innerHTML = moment().format((this.config.timeFormat === 24) ? "HH:mm" : ("hh:mm"));
		secondsWrapper.innerHTML = moment().format("ss");

MichMich, 2016-04-13:
If you could built a simple test page with moment.js, and confirm that it's an issue with the moment.js lib, you could open a issue in their repository. This way you can help them solve the issue.

Is it ok if I close this issue for now?

burnsintx, 2016-04-13:
Sorry for the late response.  Drank quite a few beers and didn't get a
chance to have a look.  I think you are right though, so go ahead and close
and I'll run some tests.  Thanks for all your hard work, this will be
great.  I'll share my final product on the reddit post when I'm finished.

closed 2016-04-13 by MichMich