MagicMirrorOrg/MagicMirror #160: [Question] Is there anyway to run v2 branch natively using nginx?

delize, 2016-04-12:
Just curious, it appears that it isn't possible, but just wanted to see if this would be something in the future, or if it is actually possible.

MichMich, 2016-04-12:
This currently isn't possible. You can run the app without electron using `start server only`, but that still includes it's own web server.

Is there any particular reason you want to use nginx?

closed 2016-04-13 by MichMich

delize, 2016-04-13:
Hey, mainly because I already have to run nginx for an embedded security camera in my mirror that goes off after I leave my house. Again, trying to keep everything simple, and the minimum amount of packages installed as necssary is helpful. Sorta goes hand in hand with the CPU Spike problem. Resources are limited, and just want to reduce any overhead if possible.