MagicMirrorOrg/MagicMirror #161: Never ending dependency problems

lastrastuff, 2016-04-12:
I keep getting to the end of the install process and it is telling me `/home/pi/MagicMirror/node_modules/electron-prebuilt/dist/electron: error while l                                             oading shared libraries: libnotify.so.4: cannot open shared object file: No such                                              file or directory` but the library changes every time I install the proper one. Am I missing a package or do I just need to get through all of these one by one?
[debug.txt](https://github.com/MichMich/MagicMirror/files/214281/debug.txt) Here is my debug file as well.

MichMich, 2016-04-12:
Thanks for your report!

Would you mind testing the [electron-quick-start](https://github.com/electron/electron-quick-start) to see if that one works?

lastrastuff, 2016-04-12:
Same error, unfortunately.

```
/home/pi/electron-quick-start/node_modules/electron-prebuilt/dist/electron error while loading shared libraries: libnotify.so.4: cannot open shared object file: No such file or directory

npm ERR! Linux 4.1.19+
npm ERR! argv "/usr/local/bin/node" "/usr/local/bin/npm" "start"
npm ERR! node v0.12.6
npm ERR! npm  v2.11.2
npm ERR! file sh
npm ERR! code ELIFECYCLE
npm ERR! errno ENOENT
npm ERR! syscall spawn
npm ERR! electron-quick-start@1.0.0 start: `electron main.js`
npm ERR! spawn ENOENT
npm ERR!
npm ERR! Failed at the electron-quick-start@1.0.0 start script 'electron main.js'.
npm ERR! This is most likely a problem with the electron-quick-start package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     electron main.js
npm ERR! You can get their info via:
npm ERR!     npm owner ls electron-quick-start
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/pi/electron-quick-start/npm-debug.log
```

MichMich, 2016-04-13:
I notice you are running a very old version of Node. (v0.12.6). Try upgrading to 5.10.1.

lastrastuff, 2016-04-13:
I successfully upgraded but now I am running into this issue. I don't even particularly know where to start with this, thanks for your help nevertheless!

```
We're ready! Run [DISPLAY=:0 npm start] from the MagicMirror directory.
pi@raspberrypi:~ $ DISPLAY=:0 npm start
npm ERR! Linux 4.1.19+
npm ERR! argv "/usr/local/bin/node" "/usr/local/bin/npm" "start"
npm ERR! node v5.10.1
npm ERR! npm  v3.8.3
npm ERR! path /home/pi/package.json
npm ERR! code ENOENT
npm ERR! errno -2
npm ERR! syscall open

npm ERR! enoent ENOENT: no such file or directory, open '/home/pi/package.json'
npm ERR! enoent ENOENT: no such file or directory, open '/home/pi/package.json'
npm ERR! enoent This is most likely not a problem with npm itself
npm ERR! enoent and is related to npm not being able to find a file.
npm ERR! enoent

npm ERR! Please include the following file with any support request:
npm ERR!     /home/pi/npm-debug.log
```

MichMich, 2016-04-13:
You are not in your MagicMirror folder. 
Try:

    cd ~/MagicMirror
    npm start

closed 2016-04-13 by MichMich

lastrastuff, 2016-04-13:
You're right! Sorry, I made that mistake earlier and posted the wrong copy. Here is what happens once I am correctly inside the directory

```
> magicmirror@2.0.0 start /home/pi/MagicMirror
> electron js/electron.js

/home/pi/MagicMirror/node_modules/electron-prebuilt/dist/electron: error while loading shared libraries: libnotify.so.4: cannot open shared object file: No such file or directory

npm ERR! Linux 4.1.19+
npm ERR! argv "/usr/local/bin/node" "/usr/local/bin/npm" "start"
npm ERR! node v5.10.1
npm ERR! npm  v3.8.3
npm ERR! file sh
npm ERR! code ELIFECYCLE
npm ERR! errno ENOENT
npm ERR! syscall spawn
npm ERR! magicmirror@2.0.0 start: `electron js/electron.js`
npm ERR! spawn ENOENT
npm ERR!
npm ERR! Failed at the magicmirror@2.0.0 start script 'electron js/electron.js'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the magicmirror package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     electron js/electron.js
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs magicmirror
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls magicmirror
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/pi/MagicMirror/npm-debug.log

```

moinahmed001, 2016-08-20:
I have this same issue as the output above!

bbreton3, 2016-08-21:
Hello, I have the same issue as above, I run node v6.4.0, have any of you guys found a way to solve the issue?

JtheB, 2016-08-21:
It looks like electron is not correctly installed, if I'm not wrong.
Maybe you could try _npm install_ again.

SimJim, 2016-09-20:
Same here...
tryed to _npm install_ again....still the same issue

MichMich, 2016-09-20:
@SimJim what version of Node are you running?

SimJim, 2016-09-20:
[image]

MichMich, 2016-09-20:
Could you give this repository a try to see if it works for you?
https://github.com/electron/electron-quick-start

SimJim, 2016-09-20:
i just tried...same error-message :/

MichMich, 2016-09-20:
Then I suggest to open an issue in that repository. It seems there is a problem with Electron.

hkapanen, 2016-09-28:
See here: https://github.com/electron/electron/issues/2472

> The only way is to make your app automatically run after the user has login, which is also how all other GUI applications do.

So I made the raspi to do desktop autologin at boot time and it fixed it for me.

BoxSo, 2016-10-13:
I ran into the same problem. Fresh install on rpi3 on latest raspbian. Is there any workaround available?

Jopyth, 2016-10-14:
In the meanwhile you could try using an older [version](https://github.com/electron/electron/releases) of electron, which does not have this bug? I.e. `npm install electron@1.3.1`.

matteos1, 2017-11-11:
here same error on electron