MagicMirrorOrg/MagicMirror #162: Display time in calendar

djsunrise19, 2016-04-12:
Hi,

actually I test the new MagicMirror² a little bit and I've seen some differences between the old and the new version in the calendar functions (time). The old version has more functionality in my opinion. For example:

old version: ManCity - PSG **in 2 hours** 
same moment on the new version: ManCity - PSG **Today**

There are a few more examples with **since 2 hours** etc. Will you implement those features in future?

Regards
Maurice

MichMich, 2016-04-13:
I'll probably add this in the future. Feel free to send a pull request.

djsunrise19, 2016-04-13:
It's the same like #166 . I tested it: Those entry's aren't whole day events and it shows no exact time.

MichMich, 2016-04-13:
Thanks for reporting ... I put it on the todo list.

MichMich, 2016-04-15:
Could you supply a test feed?

djsunrise19, 2016-04-15:
I have a nice example:

Right now on my MM² Raspi:
[image]

Right now on my MM1 Raspi:
[image]

The first event is over since 4 hours (on MM1 everything is correct). Actually I think MM² thinks all events are whole day events.

MichMich, 2016-04-15:
Yes, I understand the issue. Do you have a test .ics feed which gives you this issue?

djsunrise19, 2016-04-15:
Yes. I made a new open feed in my icloud with a couple of events:

webcal://p35-calendars.icloud.com/published/2/qsud2R2XlRhT3puEVfrzkeSd3X_QSfCMBmpgTsjVqOD2W_fyy0Y-5QskbvWw2jOQB_cPsmcjpGtcg1a3jBQ_QIS2H0XWSxjqCa8LX0dznpE

MichMich, 2016-04-15:
Thanks, I will look into it.

MichMich, 2016-04-15:
I've just pushed an update. Could you give it a try?

djsunrise19, 2016-04-15:
Looks a little bit better, but now event "Test 1" isn't there anymore. But Test 1 is from 12 to 15 o clock. So it has already begun.

MichMich, 2016-04-15:
Exactly. It is in the past, so it is filtered out.

MichMich, 2016-04-15:
It only shows future events.

djsunrise19, 2016-04-15:
Yes, thats clear. But on MM1 it was possible to see how long the event goes. Do you understand what I mean?

MichMich, 2016-04-15:
So it showed the event while it was still going ... ?

djsunrise19, 2016-04-15:
Example: Test 1 12-15 o' clock. We have 13 o'clock. So MM1 says: 2 hours left

MichMich, 2016-04-15:
Hmm, ok ... let met look at it.

MichMich, 2016-04-15:
Did you remove Test 1 from your feed?

djsunrise19, 2016-04-15:
No, I don't.

[image]

MichMich, 2016-04-15:
Ok ... let me check.

MichMich, 2016-04-15:
Ok, give it an other try. 

Make sure you set the `runningText` text property in your config to the german string. (The default is: `ends in`). So it needs to be something like: `endet am`

djsunrise19, 2016-04-15:
Thanks! Still no Test 1 event.

MichMich, 2016-04-15:
Are you sure you restarted the backend?
[image]

djsunrise19, 2016-04-15:
Yes, but I'll make a `shutdown -r 0` now.

djsunrise19, 2016-04-15:
Ah, great Michael, looks good now!

PS: In this case, in german we say `noch` :)

MichMich, 2016-04-15:
Ah, just like "Nog" in dutch! ;)

Can I close this issue?

djsunrise19, 2016-04-15:
Yes, you can :).

djsunrise19, 2016-04-15:
Hi Michael,

I've seen another constellation. As you can see, MM1 says: "Tomorrow at ***" and MM² "in 19 hours". Perhaps it's possible to use the whole event/time function from MM1 for MM²

MM²:
[image]

MM1:
[image]

Here is a test-feed for you:
webcal://p35-calendars.icloud.com/published/2/Biw06VYAVT31nDd3JqdQkTjbAzFrzU2zZjtbhH40WZ-frGU2hWaZDyG3nXBZUgk7rvBJ0x6T-SdsFyDkzkYbVP4Onr4Eb1_4BptAPzWALA4

Thanks,
Maurice

djsunrise19, 2016-04-16:
Did you see that or should I open a new issue?

MichMich, 2016-04-16:
All get back to this asap.

MichMich, 2016-04-19:
This is fixed. If a event is within 48 hours, it shows the absolute time. 
Note: This does not happen for fullday events.

I close the issue for now, if you still like some changes, feel free to let me know.

djsunrise19, 2016-04-23:
I hope it's okay that I use this "issue" again:

Today I've seen another behavior. I had an event at 16 o'clock. I looked on the mirrors at 15 o'clock:
MM1: in 1 hour
MM²: today at 16 o'clock
Is it possible to put this absolute time function in your MM² app?

Another point is, that there might be another issue. On 22.04 I had a whole day event (birthday). But today I can see this event on the MM² calendar (he says that the event is tomorrow?! I checked my event in icloud, everything is fine.

I made a feed for you:
webcal://p35-calendars.icloud.com/published/2/0gmYv_8AnTjRjR2SWtoyc7_eCWfFKOEj-ZVz7-eJ0elAZ4SDdsiCZVslxdjR-NxRhcM6N4hwTKuay0bpJSR7a9THwnMmyHEzyYFB4cpSxlA

MichMich, 2016-04-23:
`today at 16 o'clock` is an 'absolute time'. Or do you prefer a relative time?

djsunrise19, 2016-04-23:
I don't know how you solve that in MM1. A few hours before the event starts MM1 says not `today at *****` anymore but `in 3 hours` or `in 23 minutes`. I hope you understand what i mean.

BTW: Did you change anything in the calendar module? `runningText: 'noch'` doesn't work anymore. In german I see `Endet in` (like ends in).

MichMich, 2016-04-23:
Yes, but that's a personal preference. We could make it configurable. I suggest you open a separate issue for that, I'll mark it as "enhancement" and might add the feature when I have time. (Could take a while).

Also, I checked your example feed. Which event is incorrect?

About the translation: that's correct. We implemented a translation system. The strings are now translated in the language files (only the calendar for now.) See: https://github.com/MichMich/MagicMirror/issues/191

djsunrise19, 2016-04-23:
Okay I'll do that. For me, your solution in MM1 was great. 

Event "Test 8" is incorrect. It's a whole day event yesterday.

I understand. So you should update the wiki, if it doesn't work anymore :).

MichMich, 2016-04-23:
Please send a pull request to fix both the documentation and the translation if you can.

djsunrise19, 2016-04-23:
First I made an new issue for the "personal" feature. What should I do next? Create one pull request for the whole day issue and another pull request for the language problem?

MichMich, 2016-04-23:
I'll look into the incorrect Test 8 date. If you could create a pull request for the language issues, that would be awesome.

djsunrise19, 2016-04-23:
I've never done a pull request, I can try ;).

MichMich, 2016-04-23:
Looking forward to it! :)

MichMich, 2016-04-23:
Your Test 8 issue is fixed. Please update and restart :)

djsunrise19, 2016-04-23:
Thanks, I'll check that. 

Do you see my pull request?

MichMich, 2016-04-23:
Nope :(

djsunrise19, 2016-04-23:
Sorry, it's the first time. I forked the project, edit the file and made the request:

[image]

MichMich, 2016-04-23:
You are probably trying to merge an update from MM to your fork. It should be the other way around. If you go to the MM repo (not your fork), You'll see a green Pull Request button.

djsunrise19, 2016-04-23:
No I've done shit.

Robiv8, 2016-04-23:
@djsunrise19 
Ich kann dir etwas Helfen, Wenn du MM geforkt hast mußt du die Änderungen mit deinem GitHub Account machen.
Sieht bei mir so aus: Robiv8/MagicMirror dann erscheint auch der Grüne button §New pull request"

djsunrise19, 2016-04-23:
@Robiv8 
Danke dir! Ich hab's schon gefunden und Michael hat's auch schon gemerged :).

CommanderB, 2016-04-28:
In this theme I would like to suppress the exact time - its good enough to know the day it happens. So as an enhancement suggestion I created a new config variable, defaulting to true, but false for my need

`displayTime: false,`

then added the conditional in the building loop - I'm sure the conditional logic could be improved, but this works for me.

```javascript
			if (this.config.displayTime){
				if (event.fullDayEvent) {
					if (event.today) {
						timeWrapper.innerHTML = this.translate("TODAY");
					} else if (event.startDate - now < 24 * 60 * 60 * 1000 && event.startDate - now > 0) {
						timeWrapper.innerHTML = this.translate("TOMORROW");
					} else {
						timeWrapper.innerHTML =  moment(event.startDate,"x").fromNow();
					}
				} else {
					if (event.startDate >= new Date()) {
						if (event.startDate - now > 48 * 60 * 60 * 1000) {
							// if the event is no longer than 2 days away, display the absolute time.
							timeWrapper.innerHTML = moment(event.startDate,"x").fromNow();
						} else {
							timeWrapper.innerHTML = moment(event.startDate,"x").calendar();
						}
					} else {
						timeWrapper.innerHTML =  this.translate("RUNNING") + ' ' + moment(event.endDate,"x").fromNow(true);
					}
			}} else {
				if (event.today) {
					timeWrapper.innerHTML = this.translate("TODAY");
				} else if (event.startDate - now < 24 * 60 * 60 * 1000 && event.startDate - now > 0) {
					timeWrapper.innerHTML = this.translate("TOMORROW");
				} else {
					timeWrapper.innerHTML =  moment(event.startDate,"x").fromNow();
				}};

KirAsh4, 2016-05-02:
CommanderB: I would recommend starting a new post, as opposed to tagging along on this one as it's a different issue.

KirAsh4, 2016-05-11:
Hey @MichMich, you can probably close this issue now as the request from the original OP has been released. And what @CommanderB has requested above is a different issue all together.

MichMich, 2016-05-11:
Great, thanks.

commit 614e5e62 (2016-04-15): Show relative time to non full day events. Fix: #162
commit 095f08d4 (2016-04-19): Fix: #162, Possible fix for: #197
closed 2016-05-11 by MichMich