MagicMirrorOrg/MagicMirror #168: WARNING! Could not find config. Please create one.

anisimovevgeniy, 2016-04-13:
Hi,
I'm trying to use auto installer.
`│ │ ├─┬ redent@1.0.0 
│ │ │ ├─┬ indent-string@2.1.0 
│ │ │ │ └── repeating@2.0.1 
│ │ │ └── strip-indent@1.0.1 
│ │ └── trim-newlines@1.0.0 
│ ├─┬ multimatch@2.1.0 
│ │ ├── array-differ@1.0.0 
│ │ └── minimatch@3.0.0 
│ ├── normalize-selector@0.2.0 
│ ├─┬ postcss@5.0.19 
│ │ ├── js-base64@2.1.9 
│ │ ├── source-map@0.5.3 
│ │ └─┬ supports-color@3.1.2 
│ │   └── has-flag@1.0.0 
│ ├── postcss-less@0.8.0 
│ ├─┬ postcss-reporter@1.3.3 
│ │ └── lodash@4.10.0 
│ ├── postcss-resolve-nested-selector@0.1.1 
│ ├── postcss-scss@0.1.7 
│ ├─┬ postcss-selector-parser@1.3.3 
│ │ ├── flatten@1.0.2 
│ │ ├── indexes-of@1.0.1 
│ │ └── uniq@1.0.1 
│ ├── postcss-value-parser@3.3.0 
│ ├── resolve-from@2.0.0 
│ ├── specificity@0.1.5 
│ ├─┬ stylehacks@2.3.0 
│ │ ├── minimist@1.2.0 
│ │ ├─┬ read-file-stdin@0.2.1 
│ │ │ └── gather-stream@1.0.0 
│ │ └── write-file-stdout@0.0.2 
│ └── sugarss@0.1.2 
├── stylelint-config-standard@5.0.0 
├── valid-url@1.0.9 
└─┬ walk@2.3.9 
  └── foreachasync@3.0.0 

WARNING! Could not find config. Please create one.
`
And then its stops

MichMich, 2016-04-13:
`WARNING! Could not find config. Please create one.` means its started ... (Should not start automatic after installation, will fix that.)

You probably installed using SSH, so nothing happened. Try `DISPLAY=:0 nohup npm start &` from your MagicMirror folder to start it on your Pi screen.

MichMich, 2016-04-13:
If that doesn't solve the issue, please check issue: #145

closed 2016-04-13 by MichMich

codac, 2016-04-21:
Have a similar problem / same error message.
I'm using a B+ pi with Jessie. I've done the manual installation, as the automatic installation fails cause of the issue #145.
So I've executed the "noder serveronly" command and it starts but with the following output (although I do have a /config/config.js file):

> pi@raspihome ~/MagicMirror $ sudo node serveronly
Loading config ...
WARNING! Could not find config. Please create one.
Loading module helpers ...
No helper found for module: helloworld.
All module helpers loaded.
Starting server op port 8080 ...
Server started ...
Sockets connected & modules started ...

Ready to go! Please point your browser to: http://localhost:8080

MichMich, 2016-04-21:
Are you sure your MM installation is up to date? If so, could you share your config file? (please remove any personal info.)

codac, 2016-04-22:
Yes I've pulled it last week.

git pull results in:

> fatal: unable to access 'https://github.com/MichMich/MagicMirror/': Could not resolve host: github.com

This is my config file (/home/pi/MagicMirror/config/config.js):

`` 
/* Magic Mirror Config Sample
 *
 * By Michael Teeuw http://michaelteeuw.nl
 * MIT Licensed.
 */

var config = {
	port: 5901,
	language: 'de',
	timeFormat: 24,

	modules: [
		{
			module: 'alert',
                        config: {
                                effect: bouncyflip,
                                alert_effect: jelly,
                                display_time: 3.5,
                                position: center,
                                welcome_message: 'Welcome Master!'
                        }
		},
		{
			module: 'clock',
			position: 'top_left',
                        config: {
                                timeFormat: 24,
                                displaySeconds: true,
                                showPeriod: false,
                                showPeriodUpper: false
                        }
		},
		{
			module: 'calendar',
			header: 'Kalender',
			position: 'top_left',
			config: {
				calendars: [
					{
						symbol: 'calendar-check-o ',
						url: 'CAL_URL'
					}
				]
                                maximumEntries: 10,
                                maximumNumberOfDays: 365,
                                displaySymbol: true,
                                defaultSymbol: calendar,
                                maxTitleLength: 25,
                                fetchInterval: 900000,
                                animationSpeed: 2000,
                                fade: true,
                                fadePoint: 0.25,
                                loadingText: 'Einträge werden geladen...',
                                emptyCalendarText: 'Keine Einträge vorhanden...',
                                todayText: 'Heute',
                                tomorrowText: 'Morgen',
                                runningText: 'Endet in'
			}
		},
		{
			module: 'currentweather',
[... 58 lines cut ...]
                                        title: "N-TV",
                                        url: "http://www.n-tv.de/rss",
                                    }
                                ],
                                showSourceTitle: true,
                                showPublishDate: true,
                                showDescription: true,
                                reloadInterval: 900000,
                                updateInterval: 300000,
                                animationSpeed: 2000
                        }
		},
                {
                        module: 'mmm-systemtemperature',
                        position: 'top-center',
                        classes: 'small dimmed',
                        config: {
                                prependString: 'SysTemp: '
                        }
                }
	]
};

/*************** DO NOT EDIT THE LINE BELOW ***************/
if (typeof module !== 'undefined') {module.exports = config;}

MichMich, 2016-04-22:
Please update again. Last week's version is already way outdated (and probably doesn't include the fix you need.)

Also you might want to check the connectivity of your Pi, since GitHub should d be reachable.

apakrash, 2019-04-29:
I got a similar issue,

pi@raspberrypi:~/MagicMirror $ npm start

> magicmirror@2.7.1 start /home/pi/MagicMirror
> sh run-start.sh

Starting MagicMirror: v2.7.1
Loading config ...
WARNING! Could not find config file. Please create one. Starting with default configuration.
Loading module helpers ...

In the Magic Mirror folder, there was a config.js.sample, please copy/rename it as config.js

pi@raspberrypi:~/MagicMirror/config $ ls
config.js  config.js.sample

It worked fine thereafter.

sdetweil, 2019-04-29:
the automated install script copies the sample config fore you, the manual install (you doing the work)  does not, but there are instructions to do that