MagicMirrorOrg/MagicMirror #169: Repeating appointments in calendar are not shown

githubhohdorfnet, 2016-04-13:
Repeating appointments in a Google calendar (also testet Exchange 2013) are not shown by the calendar module. Can this be changed?
Thanks!

MichMich, 2016-04-13:
Could you supply an example calendar to test?

githubhohdorfnet, 2016-04-13:
`BEGIN:VCALENDAR
METHOD:PUBLISH
PRODID:Microsoft Exchange Server 2010
VERSION:2.0
X-WR-CALNAME:Kalender
BEGIN:VTIMEZONE
TZID:W. Europe Standard Time
BEGIN:STANDARD
DTSTART:16010101T030000
TZOFFSETFROM:+0200
TZOFFSETTO:+0100
RRULE:FREQ=YEARLY;INTERVAL=1;BYDAY=-1SU;BYMONTH=10
END:STANDARD
BEGIN:DAYLIGHT
DTSTART:16010101T020000
TZOFFSETFROM:+0100
TZOFFSETTO:+0200
RRULE:FREQ=YEARLY;INTERVAL=1;BYDAY=-1SU;BYMONTH=3
END:DAYLIGHT
END:VTIMEZONE
BEGIN:VEVENT
DESCRIPTION:\n
SUMMARY:Single appointment
DTSTART;TZID=W. Europe Standard Time:20160414T173000
DTEND;TZID=W. Europe Standard Time:20160414T180000
UID:F734C2F8-BE24-42FF-B554-BAA4765A7395
CLASS:PUBLIC
PRIORITY:5
DTSTAMP:20160413T141656Z
TRANSP:OPAQUE
STATUS:CONFIRMED
SEQUENCE:0
X-MICROSOFT-CDO-APPT-SEQUENCE:0
X-MICROSOFT-CDO-BUSYSTATUS:BUSY
X-MICROSOFT-CDO-INTENDEDSTATUS:BUSY
X-MICROSOFT-CDO-ALLDAYEVENT:FALSE
X-MICROSOFT-CDO-IMPORTANCE:1
X-MICROSOFT-CDO-INSTTYPE:0
X-MICROSOFT-DISALLOW-COUNTER:FALSE
END:VEVENT
BEGIN:VEVENT
DESCRIPTION:\n
RRULE:FREQ=WEEKLY;UNTIL=20161013T160000Z;INTERVAL=1;BYDAY=TH;WKST=MO
SUMMARY:Repeating appointment
DTSTART;TZID=W. Europe Standard Time:20160414T180000
DTEND;TZID=W. Europe Standard Time:20160414T183000
UID:B42FDDB0-DDA5-4A49-A826-FF5F353ED6B3
CLASS:PUBLIC
PRIORITY:5
DTSTAMP:20160413T141656Z
TRANSP:OPAQUE
STATUS:CONFIRMED
SEQUENCE:0
X-MICROSOFT-CDO-APPT-SEQUENCE:0
X-MICROSOFT-CDO-BUSYSTATUS:BUSY
X-MICROSOFT-CDO-INTENDEDSTATUS:BUSY
X-MICROSOFT-CDO-ALLDAYEVENT:FALSE
X-MICROSOFT-CDO-IMPORTANCE:1
X-MICROSOFT-CDO-INSTTYPE:1
X-MICROSOFT-DISALLOW-COUNTER:FALSE
END:VEVENT
END:VCALENDAR
`

githubhohdorfnet, 2016-04-13:
Yes, here it is:

[reachcalendar.txt](https://github.com/MichMich/MagicMirror/files/217285/reachcalendar.txt)

MichMich, 2016-04-15:
Should be fixed now. If not, please let me know.

closed 2016-04-15 by MichMich

githubhohdorfnet, 2016-04-15:
Works great!
Thanks!

commit f45a2e78 (2016-04-15): Fix repeating appointments, Fix: #169