MagicMirrorOrg/MagicMirror #173: outlook.com calendar issue

Jodyl1971, 2016-04-14:
I've tried all the variations of URLs for a public outlook.com calendar (webcal:// webcals:// http:// and https:// to the calendar.ics), but the mirror calendar only displays the status (Free, busy, tentative), not the summary text. 

My sample event is as follows in the .ics file:
BEGIN:VEVENT
UID:5f4834ab-472a-4a40-a07c-6a94a091fd6a
CLASS:PUBLIC
X-MICROSOFT-CDO-BUSYSTATUS:BUSY
TRANSP:OPAQUE
SEQUENCE:0
CREATED:20160414T203323Z
LAST-MODIFIED:20160414T203323Z
DTSTART;TZID=Eastern Standard Time:20160415T160000
DTEND;TZID=Eastern Standard Time:20160415T170000
SUMMARY:Test Event
PRIORITY:0
BEGIN:VALARM
ACTION:DISPLAY
TRIGGER:-PT15M
END:VALARM
END:VEVENT

But the calendar ist on the mirror only shows events with a status other than "Free", and only shows the status, not the summary text. The above appears as "Busy      in 13 hours" rather than "Test Event    in 13 hours".

MichMich, 2016-04-15:
Could you supply a test url of a feed with this issue?

MichMich, 2016-04-15:
[test.txt](https://github.com/MichMich/MagicMirror/files/220739/test.txt)

MichMich, 2016-04-15:
I just tried it with the test.txt file (your .ics code) above, and I get the following. 
[image]

closed 2016-04-15 by MichMich