MagicMirrorOrg/MagicMirror #176: Problems with umlauts in newsfeed (äöüß)

djsunrise19, 2016-04-15:
[image]

Problem exists directly on raspi and with my computer (`raspi_ip`:`port`)

MichMich, 2016-04-15:
In your newsfeed config, add the following property:

    encoding:'ISO-8859-1'

For more info see the [newsfeed documentation](https://github.com/MichMich/MagicMirror/tree/v2-beta/modules/default/newsfeed) and issue #126.

closed 2016-04-15 by MichMich

djsunrise19, 2016-04-15:
Thank you very much, certainly it works. I'm sorry, I have to read a little bit more before I open issues.

MichMich, 2016-04-15:
No problem. Glad I could help!

KX900, 2017-02-26:
And how to solve the same problem in compliements?
When i add this in compliments, magic mirror doesnt show the module anymore.