MagicMirrorOrg/MagicMirror #178: Please create a config file.

nickroz7, 2016-04-15:
Hey guys, I'm generally pretty new to the Raspberry Pi and coding in general. I followed the steps for configuring my raspberry pi, but I am still getting this message. Can anyone help? Thanks a bunch guys!

[image]

skibol3k, 2016-04-15:
Duplicate config/config.js.sample to config/config.js.
Modify your required settings and save.

MichMich, 2016-04-15:
Your config is probably malformed. Check your web inspector.

nickroz7, 2016-04-15:
How do I duplicate config/config.js.sample?

jferch, 2016-04-15:
cp config/config.js.sample config/config.js

MichMich, 2016-04-15:
Uh.... copy the file and rename?!

skibol3k, 2016-04-15:
Go to config folder and rename config.js.sample to config.js and modify the config file in order to make your mirror your own.

MichMich, 2016-04-15:
Preferably copy the config.js.sample. If you simply rename it, the old .sample file gets deleted and this your git folder is "dirty" which doesn't allow you to simply update using `git pull`.

But since @nickroz7 doesn't know how to copy a file, this is probably too much info ...

closed 2016-04-15 by nickroz7